meshtastic/firmware #5398: [Bug]: NMEA wpl sentences are broken now with the colorization of logs

thebentern, 2024-11-19:
### Category

Serial

### Hardware

Not Applicable

### Firmware Version

2.5.X

### Description

Reported by Marc from CalTopo. We need to prefix sentenced with \r\n.

fifieldt, 2024-11-20:
Just checking - is this a caltopo specific thing, or do they rely on output from GPS_DEBUG? If the latter, might be broken by more than just colorization.

caveman99, 2024-11-20:
> Just checking - is this a caltopo specific thing, or do they rely on output from GPS_DEBUG? If the latter, might be broken by more than just colorization.

The GPS sentences are folded into the debug log but are independent of each other. This is also only true if you select to emit the serial module output over the debug port. default behaviour is to use a different serial port..

PR #5412 (2024-11-21): Fixed NMEA sentence issue in CalTopo as well as bug with no printing all of the nodes
Closes #5398