meshtastic/firmware #5401: [Bug]: Unable to install any armhf release on kali armv7l (Raspberry Pi 0 2w)

alcor55, 2024-11-19:
### Category

Other

### Hardware

Linux Native

### Firmware Version

From 2.5.3 to 2.5.13

### Description

Same output for every release version.  Solutions?

### Relevant log output

```Shell
apt install ./meshtasticd_2.5.11.8e2a3e5_armhf.deb 
Note, selecting 'meshtasticd' instead of './meshtasticd_2.5.11.8e2a3e5_armhf.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

Unsatisfied dependencies:
 meshtasticd : Depends: libyaml-cpp0.7 but it is not installable
               Depends: libulfius2.7 but it is not installable
Error: Unable to correct problems, you have held broken packages.

-------------------

apt search libyaml-cpp0.
libyaml-cpp0.8/kali-rolling,now 0.8.0+dfsg-6+b2 armhf [installed]
  YAML parser and emitter for C++

--------------------

apt search libulfius2.7
libulfius2.7t64/kali-rolling,now 2.7.15-2.2+b1 armhf [installed,automatic]
  Web Framework for REST Applications in C
```

thebentern, 2024-11-20:
@jp-bennett any ideas?

alcor55, 2024-11-21:
Thank you for your reply. Since Kali is based on Debian and not Ubuntu, it should be compatible, right? In this case, do you have any suggestions on what I can try to install Meshtastic on Kali? Any alternative ways to obtain the missing dependencies? Thank you for your time.

closed 2024-11-21 by thebentern

jp-bennett, 2024-11-21:
> Thank you for your reply. Since Kali is based on Debian and not Ubuntu, it should be compatible, right? In this case, do you have any suggestions on what I can try to install Meshtastic on Kali? Any alternative ways to obtain the missing dependencies? Thank you for your time.

Kali is based on Debian testing, so closer to Ubuntu. The rolling aspect of Kali makes it a challenge to package for. You can compile it yourself, especially if you're on a Pi 4 or 5.

alcor55, 2024-11-22:
Thank you for the clarifications. After a long night of creating custom Kali (arm64) images for the Pi Zero 2 W and an extensive search for packages, I managed to install Meshtastic. I still don’t know if it works or not, but at least it starts.
[image]

alcor55, 2024-11-22:
I also tried to build a custom mesh firmware, but honestly unsuccessifull