meshtastic/firmware #5423: 🚀 [Feature Request]: Unification of MQTT Settings

Xaositek, 2024-11-22:
### OS

iOS, iPadOS, macOS

### Description

I'd like to suggest in the Apple programs, we look at unifying the location of similar settings, starting with MQTT. Setting up a new node and verifying configuration across two settings pages and up to 8 channel pages and result in missed elections.

Currently you must visit and verify:

- Settings > LoRa Config > Ignore MQTT or Ok to MQTT
- Settings > MQTT - verify base configuration and setup
- Channels > (Visit each channel) > Uplink and/or Downlink are selected

The ask here is to unify all MQTT related checks, configuration and opt-in/opt-out into a single settings screen.

### Participation

- [X] I am willing to pay to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GUVWAF, 2024-11-22:
"Ignore MQTT" and "OK to MQTT" were deliberately chosen to be in the LoRa config, because you control it for receiving and transmitting over LoRa. A user should not have to check a specific module configuration to control what's happening over LoRa in my opinion.

garthvh, 2024-11-23:
Being that there is already a 3.0 issue for uplink and downlink, there is not much left here

closed 2024-11-23 by garthvh