meshtastic/firmware #5438: [Bug]: Altitude keeps resetting to 0 when using Fixed Position

lagunacomputer, 2024-11-24:
### Category

Other

### Hardware

DIY

### Firmware Version

2.5.13

### Description

When using Fixed Position, and manually typing in a ## in meters in the (at least) the Android app, the Altitude resets to 0 after a few seconds.  Both in app Nodes detail section,  and in Radio Configuration -> Position.

Observed Conditions:
DIY tbeam clone with no GPS and GPS Disabled set.
using ##.##### number of digits for Long and Lat
2 digits for Altitude (ie 80)
2.5.13 fw

lagunacomputer, 2024-11-25:
This issue may have gone away w 2.5.14. I am not seeing it anymore since updating today.

fifieldt, 2024-11-25:
Thanks for letting us know!

closed 2024-11-25 by fifieldt