meshtastic/firmware #5441: How to make a hex (boot+app)

DanielCao0, 2024-11-25:
### Platform

NRF52

### Description

At present, platform only app hex , how to make a hex (app+boot), can be used for a large number of production JLINK download

fifieldt, 2024-11-25:
Something like?

bin/mergehex -m bin/s140_nrf52_7.3.0_softdevice.hex wio-sdk-wm1110/firmware.hex -o firmware.hex

DanielCao0, 2024-11-25:
> s140_nrf52_7.3.0_softdevice

PS F:\Meshtap\firmware\bin> mergehex.exe -m s140_nrf52_7.3.0_softdevice.hex ..\.pio\build\rak_wismeshtap\firmware.hex -o factory.hex
Parsing input files.
Merging file "s140_nrf52_7.3.0_softdevice.hex" into output.
Merging file "firmware.hex" into output.
Overlapping segments detected at address 155648.ERROR: The hex files cannot be merged since there are conflicts.
PS F:\Meshtap\firmware\bin>

markbirss, 2024-11-25:
https://github.com/meshtastic/firmware/blob/37da78919a679a632eea29f4b3d6d1a4cbbd84fb/bin/build-nrf52.sh#L38

DanielCao0, 2024-11-25:
I want a full hex, including  ( dfu boot+ app + s140 ),  via jlink to a empty chip

thebentern, 2024-11-25:
Meshtastic doesn't add bootloaders to the firmware hex.

closed 2024-11-29 by DanielCao0