meshtastic/firmware #5451: [Bug]: OK to MQTT flag behaves differently than documentation

ndoo, 2024-11-26:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.5.x

### Description

[Docs](https://meshtastic.org/docs/configuration/radio/lora/#ok-to-mqtt):

> This configuration only applies to Channels configured with the defaultpsk and eventpsk keys set in the Meshtastic Firmware; Channels with custom keys ignore this setting.

Code:

https://github.com/meshtastic/firmware/blob/ae4f54224ed21ca5d2a7b9c8706d74c9a6b955e5/src/mqtt/MQTT.cpp#L558

It does not do a check for those conditions, thus private i.e. non-default encryption key channels will not uplink to MQTT, which goes against the conditions/expected behaviour of the docs.

caveman99, 2024-11-26:
this is checked literally one line below the one you linked.
https://github.com/meshtastic/firmware/blob/ae4f54224ed21ca5d2a7b9c8706d74c9a6b955e5/src/mqtt/MQTT.cpp#L561C9-L565C20

closed 2024-11-26 by caveman99

ndoo, 2024-11-26:
Sorry for missing it out. For some reason, I am not seeing MQTT with a unique channel key while this flag is off, but maybe PEBKAC, I will continue testing.