meshtastic/firmware #5476: [Feature Request]: Possibility to manually apply offset for temperature, humidity and barometric pressure measurements (BME280 and other sensors)

Synthacon, 2024-11-29:
### Platform

ESP32

### Description

It would be useful to have built-in support for applying user-defined offsets to the temperature, humidity, and barometric pressure readings for the BME280 results. This sensor can't be recalibrated and humidity value drifts with the age. This option would allow users to easily adjust for systematic inaccuracies or environmental influences (e.g., heat from nearby components).

Correct for localized heating affecting temperature and humidity readings.
Adjust for known inaccuracies in factory calibration.
Enable quick fine-tuning for specific environmental conditions.

Just as example my sensor shows +0.64c and -10% for humidity and currently there is no way to adjust this.

HarukiToreda, 2024-11-29:
this is indeed a problem plaguing most BME680 sensors as well as they are not calibrated for accuracy at factory and there's no way to do so on the board itself. It must be offset by the firmware to be able to get accurate readings.

dm5tt, 2024-11-30:
They are measuring the PCB temperature and not environmental temperature. So maybe they work as designed?