meshtastic/firmware #5485: [Bug]: Considering use aht20's temperature when aht20 and bmp280 are both present

Enter-tainer, 2024-12-01:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

v2.5.14.f2ee0df

### Description

#3977 add support for ahtx0 series temperature and humidity sensor. When ahtx0 and bmp280 are both present, meshtastic will [use bmp280's temperature](https://github.com/meshtastic/firmware/pull/3977#discussion_r1617034084).

However, it seems that aht20's temperature accuracy is actually higher. In [page 8 of bmp280's data sheet](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bmp280-ds001.pdf), we can see that the temp acc is +- 0.5(or 1) degree.
[image]

While [AHT20's datasheet](https://files.seeedstudio.com/wiki/Grove-AHT20_I2C_Industrial_Grade_Temperature_and_Humidity_Sensor/AHT20-datasheet-2020-4-16.pdf) says that it has +- 0.3 typical accuray, and the max error between 0~65 is +- 0.5 degree.
[image]

So I think in this case, maybe switching to aht20 can bring better temperature result? I'm very new to embeded systems so please correct me if I get it wrong. 

cc @caveman99 @fzellini FYI

closed 2024-12-17 by garthvh

Enter-tainer, 2024-12-17:
Hi, thank you for reviewing this proposal. Could you please provide more details on why this was closed as "not planned"? I'd love to better understand your perspective and see if there's anything I can do.

garthvh, 2024-12-18:
We are going to continue to preference the better Bosch sensor