meshtastic/firmware #5486: [Feature Request]: firmware should validate private_key and public_key values match

powersjcb, 2024-12-01:
### Platform

Cross-Platform

### Description

Discovered this is not validated after typo'ing a private key value from a screenshot of my backups.

The private and public key values were still usable, but they did not match.

tgies, 2025-01-22:
Really, since public key is derived from private key, it seems to me we shouldn't allow setting the public key at all -- just regenerate it from the private key when that is set. Am I missing anything here?

garthvh, 2025-01-22:
> Really, since public key is derived from private key, it seems to me we shouldn't allow setting the public key at all -- just regenerate it from the private key when that is set. Am I missing anything here?

No

tgies, 2025-01-23:
I imagine that, besides a change to the firmware to self-set the public key when the private key is changed and no longer allow manually changing the public key, this would need corresponding changes to all clients so they no longer make it "look like" the public key can be set independently (and in fact no longer try to set it). This might be a little funky to figure out how to handle in the mobile and web app UIs (e.g. should it show a read-only pubkey field, clear it as soon as the user edits the privkey field, and then repopulate it when the configuration is pulled to the client again after the change is committed? or something else?) but I suppose some of that might be a discussion for the clients' respective issue trackers.

Then, there would maybe also need to be a corresponding documentation change to make it clear that the public key is wholly determined by the private key.

I'm willing to start putting together the PRs for this given some consensus about how it should be handled.

closed 2025-08-02 by garthvh