meshtastic/firmware #5489: [Bug]: SX1276 behavior when Transmit Power set at 1dBm

todd-herbert, 2024-12-02:
### Category

Hardware Compatibility

### Hardware

T-Lora v2 1.6

### Firmware Version

2.5.14.f2ee0df

### Description

Context: new user playing with scavenged hardware. Experiment was "listen-but-not-transmit" on 915MHz using their existing 868MHz antenna.

Region set to ANZ, transmit power set to 1dBm. Intention was to protect radio during initial reboot after setting region, where tx will briefly re-enable itself.

On reboot, critical error 3 is shown, with radiolib error code -13 (invalid output power, [minimum supposedly 2dBm](https://github.com/jgromes/RadioLib/blob/a692b72347b3ddf4ce225b0afc249c1b5a445063/src/modules/SX127x/SX1276.h#L34)).

Despite the error 3, the radio appears(?) to have continued on to transmit packets. Value of `config.lora.tx_enabled` seems to have been ignored, even after user made several attempts to toggle the setting.

___

Does anyone know what actually occurs with the radio in this situation? If an invalid output power is set, does the radio attempt to transmit at a much higher power than expected by the user? I realize this is an edge case.

### Relevant log output

```Shell
//\ E S H T /\ S T / C

  INFO  | ??:??:?? 0  Booted, wake cause 0 (boot count 1), reset_reason=reset
  DEBUG | ??:??:?? 0  Filesystem files (905216/1048576 Bytes):
  DEBUG | ??:??:?? 0   /prefs/channels.proto (57 Bytes)
  DEBUG | ??:??:?? 0   /prefs/config.proto (164 Bytes)
  DEBUG | ??:??:?? 0   /prefs/db.proto (394 Bytes)
  DEBUG | ??:??:?? 0   /prefs/module.proto (91 Bytes)
  DEBUG | ??:??:?? 0   /static/.gitkeep (0 Bytes)
  DEBUG | ??:??:?? 0   /static/Logo_Black.svg.gz (602 Bytes)
  DEBUG | ??:??:?? 0   /static/Logo_White.svg.gz (610 Bytes)
  DEBUG | ??:??:?? 0   /static/apple-touch-icon.png.gz (3164 Bytes)
  DEBUG | ??:??:?? 0   /static/favicon.ico.gz (2270 Bytes)
  DEBUG | ??:??:?? 0   /static/icon.svg.gz (852 Bytes)
  DEBUG | ??:??:?? 0   /static/index-b9vwykds.js.gz (222120 Bytes)
  DEBUG | ??:??:?? 0   /static/index-wV0rbIjU.css.gz (16573 Bytes)
  DEBUG | ??:??:?? 0   /static/index.html.gz (547 Bytes)
  DEBUG | ??:??:?? 0   /static/mapbox-gl-DY02hNYD.js.gz (391221 Bytes)
  DEBUG | ??:??:?? 0   /static/maplibre-gl-H6mDd47p.js.gz (210429 Bytes)
  DEBUG | ??:??:?? 0   /static/robots.txt.gz (42 Bytes)
  DEBUG | ??:??:?? 0   /static/site.webmanifest.gz (197 Bytes)
 [   544][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
 DEBUG | ??:??:?? 0  Power::lipoInit lipo sensor is not ready yet
  DEBUG | ??:??:?? 0  Using analog input 35 for battery level
  INFO  | ??:??:?? 0  ADCmod: ADC characterization based on reference voltage stored in eFuse
  INFO  | ??:??:?? 0  Scanning for i2c devices...
 [   587][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
 DEBUG | ??:??:?? 0  Scanning for I2C devices on port 1
  DEBUG | ??:??:?? 0  I2C device found at address 0x3c
  INFO  | ??:??:?? 0  ssd1306 display found
  INFO  | ??:??:?? 0  ssd1306 display found
  DEBUG | ??:??:?? 0  0x4 subtype probed in 2 tries 
  INFO  | ??:??:?? 0  1 I2C devices found
  DEBUG | ??:??:?? 0  acc_info = 0
  INFO  | ??:??:?? 0  S:B:3,2.5.4.8d288d5
  INFO  | ??:??:?? 0  Build timestamp: 1727744785
  DEBUG | ??:??:?? 0  Total heap: 192384
  DEBUG | ??:??:?? 0  Free heap: 157472
  DEBUG | ??:??:?? 0  Total PSRAM: 0
  DEBUG | ??:??:?? 0  Free PSRAM: 0
  DEBUG | ??:??:?? 0  NVS: UsedEntries 78, FreeEntries 552, AllEntries 630, NameSpaces 4
[... 122 lines cut ...]
  DEBUG | ??:??:?? 46 [Router]  didn't find pending packet
  DEBUG | ??:??:?? 46 [Router]  Add packet record (id= fr=0x1c to=0x1c, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=4b531681 priority=120)
  DEBUG | ??:??:?? 46 [Router]  handleReceived(REMOTE) (id= fr=0x1c to=0x1c, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=4b531681 priority=120)
  DEBUG | ??:??:?? 46 [Router]  Module 'routing' wantsPacket=1
  INFO  | ??:??:?? 46 [Router]  Received routing from=, id=, portnum=5, payloadlen=2
  DEBUG | ??:??:?? 46 [Router]  Routing sniffing (id= fr=0x1c to=0x1c, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=4b531681 priority=120)
  DEBUG | ??:??:?? 46 [Router]  Received a nak for , stopping retransmissions
  DEBUG | ??:??:?? 46 [Router]  Delivering rx packet (id= fr=0x1c to=0x1c, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=4b531681 priority=120)
  DEBUG | ??:??:?? 46 [Router]  Update DB node , rx_time=0
  DEBUG | ??:??:?? 46 [Router]  Forwarding to phone (id= fr=0x1c to=0x1c, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=4b531681 priority=120)
  DEBUG | ??:??:?? 46 [Router]  Module 'routing' considered
  DEBUG | ??:??:?? 61 [Power]  Battery: usbPower=1, isCharging=1, batMv=4356, batPct=100
  DEBUG | ??:??:?? 81 [Power]  Battery: usbPower=1, isCharging=1, batMv=4356, batPct=100
  DEBUG | ??:??:?? 101 [Power]  Battery: usbPower=1, isCharging=1, batMv=4335, batPct=100
  DEBUG | ??:??:?? 121 [Power]  Battery: usbPower=1, isCharging=1, batMv=4333, batPct=100
  DEBUG | ??:??:?? 141 [Power]  Battery: usbPower=1, isCharging=1, batMv=4320, batPct=100
  DEBUG | ??:??:?? 161 [Power]  Battery: usbPower=1, isCharging=1, batMv=4247, batPct=100
  DEBUG | ??:??:?? 181 [Power]  Battery: usbPower=1, isCharging=1, batMv=4260, batPct=100
  DEBUG | ??:??:?? 201 [Power]  Battery: usbPower=1, isCharging=1, batMv=4272, batPct=100
  DEBUG | ??:??:?? 221 [Power]  Battery: usbPower=1, isCharging=1, batMv=4249, batPct=100
  DEBUG | ??:??:?? 241 [Power]  Battery: usbPower=1, isCharging=1, batMv=4264, batPct=100
  DEBUG | ??:??:?? 261 [Power]  Battery: usbPower=1, isCharging=1, batMv=4275, batPct=100
  DEBUG | ??:??:?? 281 [Power]  Battery: usbPower=1, isCharging=1, batMv=4285, batPct=100
  DEBUG | ??:??:?? 301 [Power]  Battery: usbPower=1, isCharging=1, batMv=4307, batPct=100
```

fifieldt, 2024-12-07:
Critical Error 3 "no radio" - meshtastic_CriticalErrorCode_NO_RADIO is raised in main.cpp when there's no radio interface (!rIf) .. as a result, there's no interface added to the router so it can't route packets.

Since there's no radio interface, it can't TX via radio, but various things will still try (and fail). Since in theory it could TX via MQTT, or to the phone.

We have https://github.com/meshtastic/firmware/issues/3400 - to try and handle RadioLib error messages better.

fifieldt, 2024-12-07:
FYI, where it says error=4 in the logs, that's  meshtastic_Routing_Error_NO_INTERFACE

closed 2024-12-08 by todd-herbert