meshtastic/firmware #5491: [Bug]: RAK4631 not responding

cracky22, 2024-12-02:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.5.15.79da236

### Description

I've already tried restarting, resetting and even re-flashing, but my RAK4631 with nrf52 chip doesn't hold 80 nodes. I can definitely identify the problem in connection with the nodes because I have created a channel where there are only 8 nodes and the device stays online there. 

With over 80 nodes the rak crashes from time to time (between 6 and 9 hours) and the green LED lights up continuously.

What can I do??

SimbimChimbetov, 2024-12-04:
This problem only on 2.5.15?

cracky22, 2024-12-04:
> This problem only on 2.5.15?

No, unfortunately not. I have already tried older fw versions and always encountered this problem. Do you know a working fw? Actually it works fine if I just change the channel Our place but I do not use the ones from the city

SimbimChimbetov, 2024-12-04:
I'm not sure but maybe I have the same problem with v 2.5.11
I recently updated a Node on a Mountain with ober 100 nodes in reach, few days ago I lost Signal, last signal showed 88% Battery and the weather was fine.
Because it is really hard to access and I don't had the time to go there, I can't confirm, maybe it is just stolen.
I hope I have the time this Weekend.

cracky22, 2024-12-04:
It would be interesting to know if it is the same for other nrf52 devices. I have a t1000-e but it works there. Could it be due to the different storage options such as ram and eeprom?

markbirss, 2024-12-04:
> It would be interesting to know if it is the same for other nrf52 devices. I have a t1000-e but it works there. Could it be due to the different storage options such as ram and eeprom?

All nrf52 devices have only 28kb of littlefs storage available where preferences, ble parings and nodedb are stored (the littlefs block size will also waste some space)

As a measure to prevent this issue the nodedb size was reduced from 100 to 80 recently

[image]

you can confirm the size of the nodedb file running this test fw file

(https://discord.com/channels/867578229534359593/919642584480112750/1305904252626927729)

use "list-files-s140_nrf52_611_softdevice-1.0.0.4265ae9.uf2" for rak6431 
the other for seeed boards with newer SoftDevice

cracky22, 2024-12-04:
> > It would be interesting to know if it is the same for other nrf52 devices. I have a t1000-e but it works there. Could it be due to the different storage options such as ram and eeprom?
> 
> All nrf52 devices have only 28kb of littlefs storage available where preferences, ble parings and nodedb are stored (the littlefs block size will also waste some space)
> 
> As a measure to prevent this issue the nodedb (db.proto) size was reduced from 100 to 80 recently
> 
> https://github.com/meshtastic/firmware/pull/5346
> 
> [image]
> 
> you can confirm the size of the nodedb file running this test fw file
> 
> [image]
> 
> (https://discord.com/channels/867578229534359593/919642584480112750/1305904252626927729)
> 
> use "list-files-s140_nrf52_611_softdevice-1.0.0.4265ae9.uf2" for rak6431 
> the other for seeed boards with newer SoftDevice
> 
> if you are able to share the output this could help understand the issue further
> 

I can't even open the uf2, can you put it here?

cracky22, 2024-12-04:
So that means I connect the RAK to my computer and send you the logs?

thebentern, 2024-12-04:
[list-files-s140_nrf52_611_softdevice-1.0.0.4265ae9(1).uf2.zip](https://github.com/user-attachments/files/18009109/list-files-s140_nrf52_611_softdevice-1.0.0.4265ae9.1.uf2.zip)

cracky22, 2024-12-04:
> So that means I connect the RAK to my computer and send you the logs?

.

thebentern, 2024-12-04:
I have had rotten luck reproducing this issue so far, but today I am trying an all day run of a RAK board connected to msh/US against MQTT (client proxy) to see if it triggers at all for me. >140 nodes witnessed so far. 
[image]

cracky22, 2024-12-05:
> I have had rotten luck reproducing this issue so far, but today I am trying an all day run of a RAK board connected to msh/US against MQTT (client proxy) to see if it triggers at all for me. >140 nodes witnessed so far. 
> [image]
> 

Hi, how do you get this output? Is there a tool that can log and save all the important information?

cracky22, 2024-12-05:
Or is it just serial?

thebentern, 2024-12-05:
> Or is it just serial?

It is just serial logs. I like to use tio (https://github.com/tio/tio) because it will re-attach to the device in the case of a failure or reboot.

I ran the RAK node for about 7 hours yesterday on the msh/US topic and picked up over 600 nodes with no crashes or failures. To rule out any issues with file corruption problems, have you tried a factory reset (or even just nodedb rese)?

cracky22, 2024-12-06:
> It is just serial logs. I like to use tio (https://github.com/tio/tio) because it will re-attach to the device in the case of a failure or reboot.
>
> I ran the RAK node for about 7 hours yesterday on [...]

How did you get 600 nodes? We're not talking about MQTT, are we?

garthvh, 2024-12-06:
> How did you get 600 nodes? We're not talking about MQTT, are we?

Yes, that is how you get to 600 nodes quickly, the topics are for mqtt

cracky22, 2024-12-07:
[image]
@garthvh

cracky22, 2024-12-07:
[meshtastic-log-2024-12-07T11-01-23.989Z.log](https://github.com/user-attachments/files/18048068/meshtastic-log-2024-12-07T11-01-23.989Z.log)

cracky22, 2024-12-07:
[image]

cracky22, 2024-12-10:
I know that this is unnecessary, but it would be possible if there was an option in the firmware to query how much "memory" is available on the board and that could also be displayed in the Android app in a small graphic/text

markbirss, 2024-12-10:
> I know that this is unnecessary, but it would be possible if there was an option in the firmware to query how much "memory" is available on the board and that could also be displayed in the Android app in a small graphic/text

the littlefs support specific for nrf52 dont currently have function to get free space
you could look at adding a android app feature request on the app for free memory
https://github.com/meshtastic/Meshtastic-Android/issues

cracky22, 2024-12-11:
Ok, I can do that. But what about my problem with the RAK? What can I do or how can I debug further

markbirss, 2024-12-11:
> Ok, I can do that. But what about my problem with the RAK? What can I do or how can I debug further

Ok, are you able to capture log as the reboot/crash occur at all?
Listing of files still show db.proto size after crashed ? (or it this the already provided listing ?)

cracky22, 2024-12-29:
@tavdog hi, I just discovered your commit #5670 and updated it straight away - I hope this solves my problems with the RAK. Is there a possibility that the watchdog, for example, has a fixed memory area in which it logs data? would be very useful if my RAK crashes again. Unfortunately, I can't have it connected to the PC 24/7 and save the logs

tavdog, 2024-12-29:
> @tavdog hi, I just discovered your commit #5670 and updated it straight away - I hope this solves my problems with the RAK. Is there a possibility that the watchdog, for example, has a fixed memory [...]

My fix only formats the filesystem when lfs assert is triggered. I don't think it will help your issue and if it does it will probably result is a wiped state.

analogman2, 2024-12-29:
I am experiencing the same or similar issue on a RAK Wisblock 4631. The node crashed after a week and the last time I checked >150 nodes. I got to this issue from this issue [https://github.com/meshtastic/firmware/issues/5648](https://github.com/meshtastic/firmware/issues/5648)

> DEBUG | ??:??:?? 1 Filesystem files:
> DEBUG | ??:??:?? 1  prefs (directory)
> DEBUG | ??:??:?? 1  channels.proto (147 Bytes)
> DEBUG | ??:??:?? 2  module.proto (118 Bytes)
> DEBUG | ??:??:?? 2  config.proto (169 Bytes)
> DEBUG | ??:??:?? 2  db.proto.tmp (27377 Bytes)
> DEBUG | ??:??:?? 2  adafruit (directory)
> DEBUG | ??:??:?? 2  bond_prph (directory)
> DEBUG | ??:??:?? 2  bond_cntr (directory)
> DEBUG | ??:??:?? 2 Power::lipoInit lipo sensor is not ready yet
> DEBUG | ??:??:?? 2 Use analog input 5 for battery level
> INFO | ??:??:?? 2 Scan for i2c devices
> DEBUG | ??:??:?? 2 Scan for I2C devices on port 1
> INFO | ??:??:?? 2 No I2C devices found
> DEBUG | ??:??:?? 2 acc_info = 0
> INFO | ??:??:?? 2 S:B:9,2.5.16.f81d3b0
> INFO | ??:??:?? 2 Build timestamp: 1733662250
> DEBUG | ??:??:?? 2 Reset reason: 0x0
> DEBUG | ??:??:?? 2 Set random seed 1102827228
> INFO | ??:??:?? 2 Init NodeDB
> ERROR | ??:??:?? 2 Could not open / read /prefs/db.proto
> WARN | ??:??:?? 2 Devicestate 0 is old, discard
> INFO | ??:??:?? 2 Install default DeviceState
> DEBUG | ??:??:?? 2 Initial packet id 913518860
> DEBUG | ??:??:?? 2 Partially randomized packet id 544149773
> DEBUG | ??:??:?? 2 Use nodenum 0x9e76f91f
> INFO | ??:??:?? 2 Load /prefs/config.proto
> INFO | ??:??:?? 2 Loaded /prefs/config.proto successfully
> INFO | ??:??:?? 2 Loaded saved config version 23
> INFO | ??:??:?? 2 Load /prefs/module.proto
> INFO | ??:??:?? 2 Loaded /prefs/module.proto successfully
> INFO | ??:??:?? 2 Loaded saved moduleConfig version 23
> INFO | ??:??:?? 2 Load /prefs/channels.proto
> INFO | ??:??:?? 2 Loaded /prefs/channels.proto successfully
> INFO | ??:??:?? 2 Loaded saved channelFile version 23
> DEBUG | ??:??:?? 2 cleanupMeshDB purged 0 entries
> DEBUG | ??:??:?? 2 Use nodenum 0x9e76f91f
> INFO | ??:??:?? 2 Adding node to database with 1 nodes and 153976 bytes free!
> DEBUG | ??:??:?? 2 Expand short PSK #1
> INFO | ??:??:?? 2 Wanted region 1, using US
> INFO | ??:??:?? 2 Save /prefs/db.proto
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> lfs warn:314: No more free space 224
> ERROR | ??:??:?? 2 Error: can't encode protobuf io error
> ERROR | ??:??:?? 2 LFS assert: head >= 2 && head <= lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: head >= 2 && head <= lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: head >= 2 && head <= lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: head >= 2 && head <= lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: head >= 2 && head <= lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: head >= 2 && head <= lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: head >= 2 && head <= lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: head >= 2 && head <= lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: head >= 2 && head <= lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: head >= 2 && head <= lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: head >= 2 && head <= lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: head >= 2 && head <= lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: head >= 2 && head <= lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: head >= 2 && head <= lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: block < lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: head >= 2 && head <= lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: head >= 2 && head <= lfs->cfg->block_count
> ERROR | ??:??:?? 2 LFS assert: head >= 2 && head <= lfs->cfg->block_count #75ish more times
> ERROR | ??:??:?? 2 LFS asser

cracky22, 2024-12-29:
> I am experiencing the same or similar issue on a RAK Wisblock 4631. The node crashed after a week and the last time I checked >150 nodes. I got to this issue from this issue #5648
@analogman2 sounds interesting. looks very similar to my problem. how did you get the error log? My problem is that I can hardly leave my computer plugged in the RAK and running for 9-10 hours

analogman2, 2024-12-29:
I had to bring my solar node inside and plug it into my desktop. You can use the serial monitor in a Chromium browser on this site to access serial comms: https://flasher.meshtastic.org/#

I forgot to mention that this is the second time I have experienced this issue on 2.5.16, both times I ran nrf_erase before flashing. Upgrading to 2.5.18 today.

cracky22, 2024-12-29:
yes, I know the flasher or the serial monitor. The only problem is that I can't leave a device running until the bug occurs, it takes far too long. Maybe when I have time I'll give up my laptop and let it run for a week...
I saw that your bug occurs on 2.5.16. Until 6 hours ago I was also on 2.5.16 :)

analogman2, 2024-12-29:
There is a difference between the bugs we are each experiencing. Once my issue occurs (storage full) it is persistent and the Rak device will never boot until I nrf_erase and reflash.

tjhoff, 2025-01-02:
I'm encountering something that sounds similar ((indefinite hanging with `LFS assert: block < lfs->cfg->block_count`) on a RAK4631 on `2.5.18.89ebafc`, after also seeing something similar on `2.5.15.79da236` and a custom firmware for the R1 (`firmware-r1-2.5.14`). The only way I've been able to fix it is to using the nrf_erase utility and reinstalling, at which point it works for <24hrs. I'm in a fairly dense mesh (probably ~150 nodes). 

This is a new flavor of crash for me  - while the serial logs were the similar in the older versions, they were causing boot loops instead. In all cases, this node was reinitialized after being erased and flashed. 

Potentially one interesting usage point is I'm using a non-default primary channel, then the primary Long/Fast channel in slot 1, then another private channel in slot 2. 

Here are the [logs](https://github.com/user-attachments/files/18294949/logs.txt) for the latest unresponsive hardware, but I don't have a crash log.

tavdog, 2025-01-02:
> I'm encountering something that sounds similar (indefinite hanging with `LFS assert: block < lfs->cfg->block_count`) on a RAK4631 on `2.5.18.89ebafc`, after also seeing something similar on `2.5.15.79da236` and a custom firmware for the R1 (`firmware-r1-2.5.14`). The only way I've been able to fix it is to using the nrf_erase utility and reinstalling, at which point it works for <24hrs. I'm in a fairly dense mesh (probably ~150 nodes).
> 
> This is a new flavor of crash for me - while the serial logs were similar in the older versions, they were causing boot loops instead. In all cases, this node was reinitialized after being erased and flashed.
> 
> Potentially one interesting usage point is I'm using a non-default primary channel, then the primary Long/Fast channel in slot 1, then another private channel in slot 2.
> 
> Here are the [logs](https://github.com/user-attachments/files/18294949/logs.txt) for the latest unresponsive hardware, but I don't have a crash log.

the last line in your log is the system trying to format the filesystem.  I think the large mesh is the problem here. I only got my weird boot loop after leaving my node in a very busy area and having the node db fill up I'm guessing.

tjhoff, 2025-01-02:
Interesting. I wonder if there's additional metadata that was added per node after the 80 node limit for the NRF chips was implemented, increasing the size of the storage above that 28KB, or having multiple channels impacts the logic around the 80 node limit. I'll try and get logs leading up to a crash!

cracky22, 2025-01-02:
> I'm encountering something that sounds similar (indefinite hanging with `LFS assert: block < lfs->cfg->block_count`) on a RAK4631 on `2.5.18.89ebafc`, after also seeing something similar on [...]

But why does the board reset or bootlop? The watchdog should kick in and solve the problem. Or?

cracky22, 2025-01-02:
Does the firmware for nrf52 (.uf2 files) include the webUI?

tavdog, 2025-01-02:
> But why does the board reset or bootlop? The watchdog should kick in and solve the problem. Or?

lfs assert is a problem with the spiffs filesystem.  a simple reboot doesn't fix the issue because the flash is corrupted.

andrepcg, 2025-01-04:
I'm also experiencing some issues with a RAK4631 (i have 2 and one is acting weird).

List of nodes:
- A (esp32), connected to wifi and MQTT
- B (RAK4631), node connected to a Raspberry Pi that I can SSH into. Firmware: `2.5.18.89ebafc"`
- C (RAK4631), purely remote node (solar). Firmware: `2.5.18.89ebafc"`

`A` can see `C` and somewhat see `B`. `B` can see `C`.

**Looking at the meshtastic map, node `B` stops being updated after a while (some random time).** If I ssh into the raspberry and leave a serial connection opened with `--noproto`, it never stops responding and the "last updated" info in the map is always fresh.

It also seems that triggering any serial connection with the board makes it soft reset and it starts responding again and the map info gets instantaneously updated (by talking to the `A` node).

### 1. Requesting telemetry for node `B` on node `A`:
```
meshtastic --request-telemetry --dest '!5570b7aa'
Connected to radio
Sending device_metrics telemetry request to !5570b7aa on channelIndex:0 (this could take a while)
Aborting due to: Timed out waiting for telemetry
```

### 2. Node `B` logs right after starting `--noproto`:
```
~ $ meshtastic --noproto
WARNING file:mesh_interface.py _sendToRadio line:1117 Not sending packet because protocol use is disabled by noProto
Connected to radio
Mv=4275, batPct=100
DEBUG | 13:58:21 165 [Power] Battery: usbPower=1, isCharging=1, batMv=4277, batPct=100
ERROR | 18:39:17 189821 [RadioIf] Ignore received packet due to error=-7
DEBUG | 18:39:17 189821 [RadioIf] Packet RX (noise?) : 673ms
INFO  | 18:39:17 189822 [NodeInfo] Send our nodeinfo to mesh (wantReplies=0)
DEBUG | 18:39:17 189822 [NodeInfo] cancelSending id=0x2afbc70d, removed=0
INFO  | 18:39:17 189822 [NodeInfo] Send owner !5570b7aa/RUB1-02########/RB02
DEBUG | 18:39:17 189822 [NodeInfo] Partially randomized packet id 3399141136
DEBUG | 18:39:17 189822 [NodeInfo] Update DB node 0x5570b7aa, rx_time=1736015957
DEBUG | 18:39:17 189822 [NodeInfo] handleReceived(LOCAL) (id=0xca9ac710 fr=0x5570b7aa to=0xffffffff, WantAck=0, HopLim=4 Ch=0x0 Portnum=4 rxtime=1736015957 priority=10)
DEBUG | 18:39:17 189822 [NodeInfo] No modules interested in portnum=4, src=LOCAL
DEBUG | 18:39:17 189822 [NodeInfo] Add packet record (id=0xca9ac710 fr=0x5570b7aa to=0xffffffff, WantAck=0, HopLim=4 Ch=0x0 Portnum=4 rxtime=1736015957 priority=10)
DEBUG | 18:39:17 189822 [NodeInfo] Expand short PSK #1
DEBUG | 18:39:17 189822 [NodeInfo] Use AES128 key!
DEBUG | 18:39:17 189822 [NodeInfo] enqueue for send (id=0xca9ac710 fr=0x5570b7aa to=0xffffffff, WantAck=0, HopLim=4 Ch=0x8 encrypted len=100 rxtime=1736015957 hopStart=4 priority=10)
DEBUG | 18:39:17 189822 [NodeInfo] txGood=4,txRelay=1,rxGood=10,rxBad=8
INFO  | 18:39:17 189822 [Position] Send pos@0:6 to mesh (wantReplies=0)
DEBUG | 18:39:17 189822 [Position] cancelSending id=0x4fe0cf0f, removed=0
DEBUG | 18:39:17 189822 [Position] Send location with precision 16
INFO  | 18:39:17 189822 [Position] Strip low RTCQuality (2) time from position
INFO  | 18:39:17 189822 [Position] Position reply: time=0 lat=40196xxxx lon=-8418xxxx
DEBUG | 18:39:17 189822 [Position] Partially randomized packet id 3745994513
```

### 3. Requesting telemetry for node `B` on node `A` after the step above:
```
meshtastic --request-telemetry --dest '!5570b7aa'
Connected to radio
Sending device_metrics telemetry request to !5570b7aa on channelIndex:0 (this could take a while)
Telemetry received:
Battery level: 101.00%
Voltage: 4.28 V
Total channel utilization: 3.15%
Transmit air utilization: 0.15%
Uptime: 189882 s
```

The same problem was occurring when node `B` was on version 2.5.16. **If I leave `--noproto` running on node `B` (by using the RPi), it never stops responding!**

cracky22, 2025-01-05:
My T1000-E with the latest firmware also went into a bootloop today 😭😭

tavdog, 2025-01-05:
> My T1000-E with the latest firmware also went into a bootloop today 😭😭
> 
> Edit: how can we help you to solve the problem? @garthvh @thebentern

Put it on serial and wait until it crashes to capture the serial output of a crash.

cracky22, 2025-01-06:
> > My T1000-E with the latest firmware also went into a bootloop today 😭😭
> > Edit: how can we help you to solve the problem? @garthvh @thebentern
> 
> Put it on serial and wait until it crashes to capture the serial output of a crash.
[image]

cracky22, 2025-01-06:
I always have this in the logs

`WARN | 10:32:31 720 [RadioIf] Can not send yet, busyRx`

cracky22, 2025-01-06:
if i am not connected via ble:
```
WARN | 11:44:17 767 [GPS] GPS Buffer full with 1023 bytes waiting. Flush to avoid corruption
WARN | 11:44:40 790 [Router] ToPhone queue is full, drop packet
WARN | 11:44:49 799 [RadioIf] Rate limit portnum 3
```

but if i connect:

```
WARN | 11:44:49 800 [RadioIf] Can not send yet, busyRx
WARN | 11:44:49 800 [RadioIf] Can not send yet, busyRx
WARN | 11:44:49 800 [RadioIf] Can not send yet, busyRx
WARN | 11:44:49 800 [RadioIf] Can not send yet, busyRx
WARN | 11:44:49 800 [RadioIf] Can not send yet, busyRx
WARN | 11:44:50 800 [RadioIf] Can not send yet, busyRx
ERROR | 11:44:50 739 [RadioIf]  Hardware Failure! busyTx for more than 60s
ERROR | 11:44:50 739 [RadioIf]  NOTE! Record critical error 8 at src/mesh/RadioLibInterface.cpp:100
```

tavdog, 2025-01-06:
> if i am not connected via ble:
> 
> ```
> WARN | 11:44:17 767 [GPS] GPS Buffer full with 1023 bytes waiting. Flush to avoid corruption
> WARN | 11:44:40 790 [Router] ToPhone queue is full, drop packet
> WARN | 11:44:49 799 [RadioIf] Rate limit portnum 3
> ```
> 
> but if i connect:
> 
> ```
> WARN | 11:44:50 800 [RadioIf] Can not send yet, busyRx
> ERROR | 11:44:50 739 [RadioIf]  Hardware Failure! busyTx for more than 60s
> ERROR | 11:44:50 739 [RadioIf]  NOTE! Record critical error 8 at src/mesh/RadioLibInterface.cpp:100
> ```

I have no idea why being connected to BT would cause a hardware failure in the Lora chip.  Can you revert to older firmware and see if it's firmware related and not an actual hardware failure ?

tjhoff, 2025-01-07:
Just to chime in here - this theory around bluetooth being involved in this issue would align with some of the behavior I'm seeing. I've had a node running for 4 days with no issues on 5.15 (one of the versions I was seeing issues in before), with the only difference being I've been connected to another node over LAN.

This might also explain the relative infrequency of people reporting this issue, as many NRF52 chips in my area appear to be solar nodes and have many days of uptime.

cracky22, 2025-01-08:
@JimTheCowboy hi, unfortunately this problem isn't fixed until now. You can help the devs and send logs where the error occurs

c0dexter, 2025-01-09:
I had the same issue, but I wasn't able to gather logs because after connecting RAK the serial console log didn't display any data

cracky22, 2025-01-09:
@c0dexter Do you have serial logging enabled?

c0dexter, 2025-01-09:
yes, but it was completely crashed - I used 'Open Serial Monitor' on https://flasher.meshtastic.org/ - but I didn't see any logs.

BTW. My friend probably have a good hint what is potentially wrong. This is his notice:
"I remembered the whole event of the node testing about 50m from the house on a shed in the yard and I went to check the milliammeter. After checking, when I was already walking away from the node, the phone vibrated in my pocket, it was constantly receiving a lot of outstanding messages via Bluetooth, and I was already going home - at a certain distance receiving have stopped, so I thought that the end of downloading outstanding messages. When I entered the house and checked on another phone the same node, I noticed that node was corrupted. Now I'm linking this case with issue of corrupting nodes like we're observing now"

So potential issue is related to breaking bluetooth connection during dumping messages to phone app. Maybe it required long list of messages - in above mentioned case was the same.

In my case was similar, I have outdoor node and I'm connecting to it from time to time and after the case of dumping to old-stored messages to my phone I wasn't able to connect to the node again.

JimTheCowboy, 2025-01-10:
Newb here, looking for some help. 
I want to help with a Firmware issue on the current alpha build. Does anybody know a simple version to do error loging in windows? 
I can't code or program so much of it is just gibberish to me. But I can handle a CLI if instructed in a comprehensive way...

JimTheCowboy, 2025-01-10:
> @JimTheCowboy hi, unfortunately this problem isn't fixed until now. You can help the devs and send logs where the error occurs

Hi, 
I would like to contribute to solving this issue, but I the error loging is a problem. 
Where can I find simple instructions to create logs? 

https://github.com/tio/tio is way out of my depth....

tavdog, 2025-01-10:
> Hi, I would like to contribute to solving this issue, but I the error loging is a problem. Where can I find simple instructions to create logs?
>
> https://github.com/tio/tio is way out of my depth....

Trying using flasher.meshtastic.org , there is an open serial monitor button.  It will let you watch the serial log over serial port.

JimTheCowboy, 2025-01-10:
about recreating the problem:  

As long as my node is on my person (reliable BT range) the problem does not occur at all or very rarely. 

If I leave the node in my car and move around it during the day, the problem seems to escalate somehow. 

I got a critical fault #12 while moving around 30-100 feet (10-30 meters) with some data loss (username, channels) 
I noticed this because my phone got multiple BT pairing requests although the node was paired.

The next time with the same movement I got a complete boot loop error requiring factory reset.

garthvh, 2025-01-10:
> about recreating the problem:
>
> As long as my node is on my person (reliable BT range) the problem does not occur at all or very rarely.
>
> If I leave the node in my car and move around it during the [...]

Which phone app?

JimTheCowboy, 2025-01-10:
Android 2.5.15 beta

b8b8, 2025-01-11:
> Android 2.5.15 beta

yup, same problem for me on the T1000E and a few raks. going into and out of range repeated with android eventually causes it to lock up, and i have to do a factory wipe. i am waiting for it to happen again to me to post my logs.

cracky22, 2025-01-11:
@garthvh @thebentern @markbirss here are some logs for you, pls look at the end:
```
[0m [33m WARN  [0m| 10:35:22 2126 [RadioIf] [33m Can not send yet, busyTx
[0m [31m ERROR [0m| 10:35:22 2126 [RadioIf] [31m Hardware Failure! busyTx for more than 60s
[0m [31m ERROR [0m| 10:35:22 2126 [RadioIf] [31m NOTE! Record critical error 8 at src/mesh/RadioLibInterface.cpp:100
```
[meshtastic-log-2025-01-11T09-57-08.007Z.log](https://github.com/user-attachments/files/18385391/meshtastic-log-2025-01-11T09-57-08.007Z.log)

JimTheCowboy, 2025-01-11:
Today it took me one (1) time of going out of BT Range.
Node stuck in Reboot loop. 

Node: T-Echo, FW 2.5.15.79da236 Beta
Phone: Android 13, App 2.5.15 beta

DEBUG | 11:48:21 138880 [Router] decoded message (id=0x35247922 fr=0x04 to=0xd4, WantAck=0, HopLim=2 Ch=0x2 Portnum=4 WANTRESP rxtime=1736596101 rxSNR=-16.5 rxRSSI=-117 hopStart=3)
DEBUG | 11:48:21 138880 [Router] handleReceived(REMOTE) (id=0x35247922 fr=0x04 to=0xd4, WantAck=0, HopLim=2 Ch=0x2 Portnum=4 WANTRESP rxtime=1736596101 rxSNR=-16.5 rxRSSI=-117 hopStart=3)
DEBUG | 11:48:21 138880 [Router] Module 'nodeinfo' wantsPacket=1
INFO | 11:48:21 138880 [Router] Received nodeinfo from=0x433b0304, id=0x35247922, portnum=4, payloadlen=44
INFO | 11:48:21 138880 [Router] Node database full with 80 nodes and 104904 bytes free. Erasing oldest entry
INFO | 11:48:21 138880 [Router] Adding node to database with 80 nodes and 104904 bytes free!
DEBUG | 11:48:21 138880 [Router] old user /, channel=0
DEBUG | 11:48:21 138880 [Router] Update changed=1 user Meshtastic 0304/0304, channel=2
DEBUG | 11:48:21 138880 [Router] Node status update: 55 online, 80 total
INFO | 11:48:22 138881 [Router] Save /prefs/db.proto
lfs debug:617: Bad block at 8
lfs debug:617: Bad block at 105
lfs debug:617: Bad block at 106
lfs debug:640: Relocating 8 7 to 107 7
INFO | 11:48:28 138887 [Router] BLE Disconnected, reason = 0x8
DEBUG | 11:48:28 138887 [Router] PhoneAPI::close()
ERROR | 11:48:28 138887 [Router] LFS assert: head >= 2 && head <= lfs->cfg->block_count
ERROR | 11:48:28 138887 [Router] LFS assert: block < lfs->cfg->block_count

cracky22, 2025-01-11:
@JimTheCowboy isn`t that #5648?

JimTheCowboy, 2025-01-11:
I don't know if it's https://github.com/meshtastic/firmware/issues/5648.
That's just how I got it to crash.

Waiting for my nodeDb to reach 80 again...

JimTheCowboy, 2025-01-12:
> I don't know if it's #5648. That's just how I got it to crash.
> 
> Waiting for my nodeDb to reach 80 again...

Been trying to recreate the bug since yesterday - no luck. 
NodeDB 80+ identical settings, in and out of BT Range but it won't crash.

I hope the refactor will help. 
I know of at least 6 T-echos, 1 Wismesh Pocket and one RAK 19007/4631 showing this behaviour.
Lot's of folk in our mesh dumped their T-echos because of this...

JimTheCowboy, 2025-01-12:
pounding my head:

turned off serial log.
30 minutes later: partial reset, longname/shortname erased.
other settings intact, channels saved, lora region saved, nodeDB still 80+
no reboot loop. 
no BT pairing request.

[64 more replies]

closed 2025-02-15 by fifieldt