meshtastic/firmware #5498: [Bug]: RAK4631, nRF52, mqtt message TX to local mesh rebooting

drmzperx, 2024-12-04:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.5.X

### Description

RAK4631 + ETH after fetch an mqtt channel message input json (see below), the RAK does not transmit the message to the local mesh, and the RAK reboots, but parsed! All settings are fine, Heltec v3 working good and the local mesh received the message on the primary channel via Heltec v3.

I try multiple RAK4631 with RAK19007 Base Board - RAK13800 Ethernet Module 2.5.11 -> 2.5.15 firwmare simple and ETH_GW version as well. (The simple version does not parse the input JSON as expected.) Try the erase firware, but all the same. 

Custom mqtt server injectetd input json:
{
"from": 2552625594,
"type": "sendtext",
"payload": "TEST msg"
}

Settings:

  "mqtt": {
    "enabled": true,
    "address": "sewico.com",
    "username": "xxxx",
    "password": "xxxx",
    "jsonEnabled": true,
    "root": "msh/mzpx",
    "mapReportSettings": {
      "publishIntervalSecs": 0,
      "positionPrecision": 0
    },
    "encryptionEnabled": false,
    "tlsEnabled": false,
    "proxyToClientEnabled": false,
    "mapReportingEnabled": false
  },

  "lora": {
    "usePreset": true,
    "region": "EU_868",
    "hopLimit": 6,
    "txEnabled": true,
    "txPower": 27,
    "sx126xRxBoostedGain": true,
    "modemPreset": "LONG_FAST",
    "bandwidth": 0,
    "spreadFactor": 0,
    "codingRate": 0,
    "frequencyOffset": 0.0,
    "channelNum": 0,
    "overrideDutyCycle": false,
    "overrideFrequency": 0.0,
    "ignoreIncoming": [],
    "ignoreMqtt": false
  },

  "network": {
    "ntpServer": "meshtastic.pool.ntp.org",
    "ethEnabled": true,
    "ipv4Config": {
      "ip": 956410048,
      "gateway": 16885952,
      "dns": 16885952,
      "subnet": 0
    },
    "wifiEnabled": false,
    "wifiSsid": "",
    "wifiPsk": "",
    "addressMode": "DHCP",
    "rsyslogServer": ""
  },

### Relevant log output

```Shell
DEBUG | 12:36:24 53 [Router] Incoming msg was filtered from 0x9825f9ba
INFO  | 12:36:26 55 [mqtt] JSON payload 1733315780.893 TEST BASE MQTT Node-Red test message., length 52
DEBUG | 12:36:26 55 [mqtt] Partially randomized packet id 2320905825
DEBUG | 12:36:26 55 [mqtt] Update DB node 0x9825f9ba, rx_time=1733315786
DEBUG | 12:36:26 55 [mqtt] handleReceived(LOCAL) (id=0x8a563661 fr=0xba to=0xff, WantAck=0, HopLim=6 Ch=0x0 Portnum=1 rxtime=1733315786)
DEBUG | 12:36:26 55 [mqtt] No modules interested in portnum=1, src=LOCAL
DEBUG | 12:36:26 55 [mqtt] Add packet record (id=0x8a563661 fr=0xba to=0xff, WantAck=0, HopLim=6 Ch=0x0 Portnum=1 rxtime=1733315786)
DEBUG | 12:36:26 55 [mqtt] Use AES256 key!
DEBUG | 12:36:26 55 [mqtt] MQTT onSend - Publish 
DEBUG | 12:36:26 55 [mqtt] portnum 1 message
DEBUG | 12:36:26 55 [mqtt] MQTT Publish msh/mzpx/2/e/Private/!9825f9ba, 106 bytes
DEBUG | 12:36:26 55 [mqtt] got text message of size 52
INFO  | 12:36:26 55 [mqtt] text message payload is of type plaintext
INFO  | 12:36:26 55 [mqtt] serialized json message: {"payload":{"text":"1733315780.893 TEST BASE MQTT Node-Red test message."},"id":2320905825,"timestamp":1733315786,"to":4294967295,"fr
INFO  | 12:36:26 55 [mqtt] JSON publish message to msh/mzpx/2/json/Private/!9825f9ba, 223 bytes: {"payload":{"text":"1733315780.893 TEST BASE MQTT Node-Red test message."},"id":23209058
DEBUG | 12:36:26 55 [mqtt] enqueue for send (id=0x8a563661 fr=0xba to=0xff, WantAck=0, HopLim=6 Ch=0x49 encrypted rxtime=1733315786 hopStart=6 priority=100)
DEBUG | 12:36:26 55 [mqtt] txGood=3,txRelay=1,rxGood=3,rxBad=0
DEBUG | 12:36:26 55 [mqtt] Partially randomized packet id 3310710370
DEBUG | 12:36:26 55 [mqtt] Enqueued local (id=0xc5556e62 fr=0xba to=0xba, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=8a563661 rxtime=1733315786 priority=120)
DEBUG | 12:36:26 55 [Router] Rx someone rebroadcasting for us (id=0xc5556e62 fr=0xba to=0xba, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=8a563661 rxtime=1733315786 priority=120)
DEBUG | 12:36:26 55 [Router] Didn't find pending packet
DEBUG | 12:36:26 55 [Router] Add packet record (id=0xc5556e62 fr=0xba to=0xba, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=8a563661 rxtime=1733315786 priority=120)
DEBUG | 12:36:26 55 [Router] handleReceived(REMOTE) (id=0xc5556e62 fr=0xba to=0xba, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=8a563661 rxtime=1733315786 priority=120)
DEBUG | 12:36:26 55 [Router] Module 'routing' wantsPacket=1
INFO  | 12:36:26 55 [Router] Received routing from=0x9825f9ba, id=0xc5556e62, portnum=5, payloadlen=2
DEBUG | 12:36:26 55 [Router] Routing sniffing (id=0xc5556e62 fr=0xba to=0xba, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=8a563661 rxtime=1733315786 priority=120)
DEBUG | 12:36:26 55 [Router] Received a ACK for 0x8a563661, stopping retransmissions
DEBUG | 12:36:26 55 [Router] Delivering rx packet (id=0xc5556e62 fr=0xba to=0xba, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=8a563661 rxtime=1733315786 priority=120)
DEBUG | 12:36:26 55 [Router] Update DB node 0x9825f9ba, rx_time=1733315786
DEBUG | 12:36:26 55 [Router] Forwarding to phone (id=0xc5556e62 fr=0xba to=0xba, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=8a563661 rxtime=1733315786 priority=120)
DEBUG | 12:36:26 55 [Router] Module 'routing' considered
WARN  | 12:36:26 55 [mqtt] JSON downlink received on channel not called 'mqtt' or without downlink enabled
WARNING file:stream_interface.py __reader line:187 Meshtastic serial port disconnected, disconnecting... device reports readiness to read but returned no data (device disconnected or multiple access on port?)
```

drmzperx, 2024-12-04:
> looking at your username we are part of the same mesh and some of the nodes utilises a private MQTT server to counter the zero-hop, so our mesh is flooded with MQTT messages on the default LongFast channel :)
> 
> Anyway, my nodes are dropping MQTT by default and I'm not on this private MQTT, so it shouldn't be a problem in the first place.

Your problem is completely different in this case. Here, the issue is reproducible: after parsing the mqtt channel JSON message and before the LoRa transmission - to the local mesh -, the RAK was rebooting, as show in the log. The Heltec V3 working perfectly.

drmzperx, 2024-12-10:
The new symptom (nrf bootloader erase, firmware 2.5.15) is that the RAK cannot transmit any payload (e.g. phone app message) over radio after parsing an MQTT JSON message, and the green LED continuously lights up. It seems frozen (not react any input via phone app), but it's not. After a long time, the RAK continues to work, but unfortunately not properly.

If I send a phone app message before the MQTT JSON input, the phone message is sent successfully, but the MQTT JSON input crashes (reboot) the device as mentioned earlier.

The serial output is:

```
INFO  | 14:30:28 25 [mqtt] serialized json message: {"payload":{"text":"1733841021.995 TEST BASE MQTT Node-Red test message."},"id":3555239907,"timestamp":1733841028,"to":4294967295,"fr
INFO  | 14:30:28 25 [mqtt] JSON publish message to msh/mzpx/2/json/Private/!9825f9ba, 223 bytes: {"payload":{"text":"1733841021.995 TEST BASE MQTT Node-Red test message."},"id":35552399
DEBUG | 14:30:28 25 [mqtt] enqueue for send (id=0xd3e8a7e3 fr=0xba to=0xff, WantAck=0, HopLim=6 Ch=0x49 encrypted rxtime=1733841028 hopStart=6 priority=100)
DEBUG | 14:30:28 25 [mqtt] txGood=0,txRelay=0,rxGood=0,rxBad=0
INFO  | 14:30:28 25 Tell client we have new packets 5
INFO  | 14:30:28 25 BLE notify fromNum
DEBUG | 14:30:28 25 [mqtt] Partially randomized packet id 2397532132
DEBUG | 14:30:28 25 [mqtt] Enqueued local (id=0x8ee76fe4 fr=0xba to=0xba, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=d3e8a7e3 rxtime=1733841028 priority=120)
DEBUG | 14:30:28 25 [Router] Rx someone rebroadcasting for us (id=0x8ee76fe4 fr=0xba to=0xba, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=d3e8a7e3 rxtime=1733841028 priority=120)
DEBUG | 14:30:28 25 [Router] Didn't find pending packet
DEBUG | 14:30:28 25 [Router] Add packet record (id=0x8ee76fe4 fr=0xba to=0xba, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=d3e8a7e3 rxtime=1733841028 priority=120)
DEBUG | 14:30:28 25 [Router] handleReceived(REMOTE) (id=0x8ee76fe4 fr=0xba to=0xba, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=d3e8a7e3 rxtime=1733841028 priority=120)
DEBUG | 14:30:28 25 [Router] Module 'routing' wantsPacket=1
INFO  | 14:30:28 25 [Router] Received routing from=0x9825f9ba, id=0x8ee76fe4, portnum=5, payloadlen=2
DEBUG | 14:30:28 25 [Router] Routing sniffing (id=0x8ee76fe4 fr=0xba to=0xba, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=d3e8a7e3 rxtime=1733841028 priority=120)
DEBUG | 14:30:28 25 [Router] Received a ACK for 0xd3e8a7e3, stopping retransmissions
DEBUG | 14:30:28 25 [Router] Delivering rx packet (id=0x8ee76fe4 fr=0xba to=0xba, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=d3e8a7e3 rxtime=1733841028 priority=120)
DEBUG | 14:30:28 25 [Router] Update DB node 0x9825f9ba, rx_time=1733841028
DEBUG | 14:30:28 25 [Router] Forwarding to phone (id=0x8ee76fe4 fr=0xba to=0xba, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=d3e8a7e3 rxtime=1733841028 priority=120)
DEBUG | 14:30:28 25 [Router] FromRadio=STATE_SEND_PACKETS
DEBUG | 14:30:28 25 [Router] Module 'routing' considered
INFO  | 14:30:28 25 [PowerFSM] Loss of power in Powered
DEBUG | 14:30:28 25 [PowerFSM] State: ON
INFO  | 14:30:28 25 Tell client we have new packets 6
INFO  | 14:30:28 25 BLE notify fromNum
WARN  | 14:30:29 25 [mqtt] JSON downlink received on channel not called 'mqtt' or without downlink enabled
DEBUG | 14:30:29 26 [RadioIf] FromRadio=STATE_SEND_PACKETS
DEBUG | 14:30:29 26 [RadioIf] phone downloaded packet (id=0x8ee76fe4 fr=0xba to=0xba, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=d3e8a7e3 rxtime=1733841028 priority=120)
INFO  | 14:30:54 51 [RadioIf] toRadioWriteCb data 0x2001c572, len 49
DEBUG | 14:30:54 51 [RadioIf] New ToRadio packet
DEBUG | 14:30:54 51 [RadioIf] PACKET FROM PHONE (id=0x65da47cc fr=0x00 to=0xba, WantAck=0, HopLim=6 Ch=0x0 Portnum=3 priority=10)
DEBUG | 14:30:54 51 [RadioIf] Enqueued local (id=0x65da47cc fr=0x00 to=0xba, WantAck=0, HopLim=6 Ch=0x0 Portnum=3 rxtime=1733841054 priority=10)
INFO  | 14:33:18 195 [RadioIf] toRadioWriteCb data 0x2001c572, len 49
DEBUG | 14:33:18 195 [RadioIf] New ToRadio packet
DEBUG | 14:33:18 195 [RadioIf] PACKET FROM PHONE (id=0x65da47cd fr=0x00 to=0xba, WantAck=0, HopLim=6 Ch=0x0 Portnum=3 priority=10)
DEBUG | 14:33:18 195 [RadioIf] Enqueued local (id=0x65da47cd fr=0x00 to=0xba, WantAck=0, HopLim=6 Ch=0x0 Portnum=3 rxtime=1733841198 priority=10)
INFO  | 14:35:33 330 [RadioIf] toRadioWriteCb data 0x2001c572, len 49
DEBUG | 14:35:33 330 [RadioIf] New ToRadio packet
DEBUG | 14:35:33 330 [RadioIf] PACKET FROM PHONE (id=0x65da47ce fr=0x00 to=0xba, WantAck=0, HopLim=6 Ch=0x0 Portnum=3 priority=10)
DEBUG | 14:35:33 330 [RadioIf] Enqueued local (id=0x65da47ce fr=0x00 to=0xba, WantAck=0, HopLim=6 Ch=0x0 Portnum=3 rxtime=1733841333 priority=10)
INFO  | 14:37:56 473 [RadioIf] toRadioWriteCb data 0x2001c572, len 50
DEBUG | 14:37:56 473 [RadioIf] New ToRadio packet
DEBUG | 14:37:56 473 [RadioIf] PACKET FROM PHONE (id=0x65da47cf fr=0x00 to=0xba, WantAck=0, HopLim=6 Ch=0x0 Portnum=3 priority=10)
DEBUG | 14:37:56 473 [RadioIf] Enqueued local (id=0x65da47cf fr=0x00 to=0xba, WantAck=0, HopLim=6 Ch=0x0 Portnum=3 rxtime=1733841476 priority=10)
```

garthvh, 2024-12-17:
JSON functionality in the firmware is in the process of being retired https://github.com/meshtastic/firmware/issues/5507

For this specific case you may want to try using the RAK ethernet gateway firmware which is going to retain JSON functionality

closed 2024-12-17 by garthvh

GUVWAF, 2024-12-17:
Still looks like the original JSON issue (#2804), the RAK Ethernet gateway version only uses ArduinoJson for sending packets, for parsing them it seems to still use the old library: 
https://github.com/meshtastic/firmware/blob/4edeca5f846024365bbd8855a454d42d2eae4d57/src/mqtt/MQTT.cpp#L59

drmzperx, 2024-12-18:
In firmware version 2.5.6, a patch was added for JSON parsing to MQTT: [https://github.com/meshtastic/firmware/pull/4661](https://github.com/meshtastic/firmware/pull/4661) It appears to be working well because the message is displayed correctly in the MQTT server (+/2/e/<channel name>) (transform from node-red json to protobuf and appears in mqtt). However, the RAK4631 does not transmit the radio signal.