meshtastic/firmware #5502: [Feature Request]: Fall back ft o Bluetooth when wifi connection is lost.

linker3000, 2024-12-04:
### Platform

ESP32

### Description

It would be great to have a configuration option to set that a device switches to Bluetooth if the WiFi connection is lost for over a short (user definable?) period of time. And likewise to revert to WiFi when a valid connection is restored.

This would be particularly useful for nodes kept at a base address but sometimes taken out to go mobile.

fifieldt, 2024-12-07:
Thanks for sending this in. Indeed, it would be convenient for mobile nodes.

However, is it possible to for these devices to poll for for available wifi networks when they are connected to Bluetooth? If not, it might not be possible to detect a network has re-appeared and 'resume'. My understanding is that with these devices when blueooth is on, wifi is off, and vice-versa.

garthvh, 2024-12-08:
Bluetooth and WiFi use the same radio and antenna, meshtastic can use one at a time

closed 2024-12-08 by garthvh

linker3000, 2024-12-08:
Well, that was closed quickly without much discussion or feedback.

If it's possible - which is very likely - that loss of WiFi can be detected and Bluetooth enabled that would be very useful.

If it's not possible to detect when WiFi is restored due to either-or radio use, then it would be possible to manually switch back via Bluetooth using a mobile app.

That option would remove the need to make a manual USB connection.

Please continue to consider this request - my coding skills aren't sufficient to work on this myself.

Thank you.