meshtastic/firmware #5507: [Feature Request]: Retire native JSON MQTT functionality

thebentern, 2024-12-04:
### Platform

Cross-Platform

### Description

This has been a problematic feature for several reasons:

1. It takes up precious space on NRF52 based devices which are approaching the limits of flash utilization
2. It has created potential security concerns in the past and there are likely undiscovered ones existing now.
3. It should be an application layer concern where the implementation can be less prescriptive and tailored to the use case.

Tagging @ianmcorvidae because he has some migration strategies to externalize the support while keeping things in the format they exist in currently for the firmware so that automations in Home Assistant or otherwise can continue working with a bit of extra sauce.

GUVWAF, 2024-12-04:
Agree on this, although indeed with the "requirement" that it would need to be able to easily integrate with popular home automation systems like Home Assistant and OpenHAB. Ideally it would be a native integration/binding such that the end-user doesn't have to handle this glue code and run it in an additional daemon.

pdxlocations, 2024-12-04:
It case it's useful, I've roughed out a python script for this purpose.  Maybe it's possible to use something like this as a custom addon to HA?
https://github.com/pdxlocations/mqtt-protobuf-to-json

GUVWAF, 2024-12-17:
@garth Why are we planning to retain the JSON functionality for the RAK Ethernet gateway? https://github.com/meshtastic/firmware/issues/5498#issuecomment-2549023963

Then someone would still need to maintain this (add new fields and fix bugs like the one linked), and it can only be tested on one specific hardware combo. Besides, once we have a suitable alternative it will likely be slightly different than the manually crafted JSON, so you can't use the same system with different hardware. In my opinion if we remove it from the firmware, we have to nuke it altogether.

garthvh, 2024-12-17:
> @garthvh Why are we planning to retain the JSON functionality for the RAK Ethernet gateway? [#5498 (comment)](https://github.com/meshtastic/firmware/issues/5498#issuecomment-2549023963)
> 
> Then someone would still need to maintain this (add new fields and fix bugs like the one linked), and it can only be tested on one specific hardware combo. Besides, once we have a suitable alternative it will likely be slightly different than the manually crafted JSON, so you can't use the same system with different hardware. In my opinion if we remove it from the firmware, we have to nuke it altogether.

I don't anticipate it being something that the project maintains, that device is managed by @beegee-tokyo and RAK so my thought was that he could continue to maintain it if necessary, but I am not opposed to nuking it.

beegee-tokyo, 2024-12-18:
@garthvh I am ok, _**if you replace it with something else that works.**_ 
MQTT with JSON might not be important for "messaging", but when using Meshtastic in other applications like sensor networks, control functions, it is essential, specially because you don't offer a working raw payload decoder at this point.
I can maintain MQTT until you come up with something else.

leshniak, 2024-12-19:
> It case it's useful, I've roughed out a python script for this purpose. Maybe it's possible to use something like this as a custom addon to HA? https://github.com/pdxlocations/mqtt-protobuf-to-json

I've started a similar project based on node.js a month ago (for private use at the time). Yours suits better for HA.

beegee-tokyo, 2024-12-20:
@leshniak 
Any plans to add your project to NodeRED official packages? 
I am struggling to add your meshtastic-protobuf-decoder to my NodeRED installation.

fifieldt, 2024-12-22:
@broglep  is working on a home assistant integration at https://github.com/broglep/homeassistant-meshtastic that looks like it won't need json+mqtt

pdxlocations, 2024-12-28:
An HA integration using protos and MQTT: https://github.com/kvj/hass_Mtastic_MQTT

garthvh, 2026-09-03:
Closing: native JSON MQTT support was retired from firmware in PR #10152 (commit f86cb7781), which removed the JSON.cpp/JSONValue.cpp codecs from the embedded build while keeping JSON serialization for Meshtasticd (Linux native) behind an ARCH_PORTDUINO guard. That is the compromise reached in this thread. Shipped in the 2.8 alpha (v2.8.0.47db0e3).

closed 2026-09-03 by garthvh