meshtastic/firmware #5521: [Bug]: Gps not present/ GPS enable  not working anymore on Faketec Node  since firmware version 2.5.15

lupusworax, 2024-12-07:
### Category

Other

### Hardware

Other

### Firmware Version

2.5.15.79da236

### Description

Faketec Node NRF52 dev board with HT RA-62 ( new variant that finally works flawless!)  Gps is not recognized anymore ( GPS not present) also GPS enable on GPIO pin 24 not working anymore.
Reverting back to 2.5.14 and it works again.

lupusworax, 2024-12-07:
This was tested on 4  builds of the exact same kind and used hardware I made with the exact same outcome as described above.

fifieldt, 2024-12-07:
Can you please provide logs of the early startup, perhaps using the Serial Console on https://flasher.meshtastic.org ?

lupusworax, 2024-12-07:
Log output under 2.5.14 were everything is working normal: 
```
All267
Info82
Debug178
Warn2
Error3
Disconnect

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 4 Filesystem files:

DEBUG | ??:??:?? 4  prefs (directory)

DEBUG | ??:??:?? 4  config.proto (183 Bytes)

DEBUG | ??:??:?? 4  module.proto (95 Bytes)

DEBUG | ??:??:?? 4  channels.proto (110 Bytes)

DEBUG | ??:??:?? 4  db.proto (8447 Bytes)

DEBUG | ??:??:?? 4  adafruit (directory)

DEBUG | ??:??:?? 4  bond_prph (directory)

DEBUG | ??:??:?? 4  bond_cntr (directory)

DEBUG | ??:??:?? 4 Power::lipoInit lipo sensor is not ready yet

DEBUG | ??:??:?? 4 Use analog input 31 for battery level

INFO | ??:??:?? 4 Scan for i2c devices

DEBUG | ??:??:?? 4 Scan for I2C devices on port 1

DEBUG | ??:??:?? 4 I2C device found at address 0x3c

INFO | ??:??:?? 4 ssd1306 display found

INFO | ??:??:?? 4 ssd1306 display found

DEBUG | ??:??:?? 4 0x4 subtype probed in 2 tries

INFO | ??:??:?? 4 1 I2C devices found

DEBUG | ??:??:?? 4 acc_info = 0

INFO | ??:??:?? 4 S:B:63,2.5.14.f2ee0df0

DEBUG | ??:??:?? 4 Reset reason: 0x0

DEBUG | ??:??:?? 4 Set random seed 1018213078

INFO | ??:??:?? 4 Init NodeDB

INFO | ??:??:?? 4 Load /prefs/db.proto

INFO | ??:??:?? 4 Loaded /prefs/db.proto successfully

INFO | ??:??:?? 4 Loaded saved devicestate version 23, with nodecount: 76

INFO | ??:??:?? 4 Load /prefs/config.proto

INFO | ??:??:?? 4 Loaded /prefs/config.proto successfully

INFO | ??:??:?? 4 Loaded saved config version 23

INFO | ??:??:?? 4 Load /prefs/module.proto
[... 446 lines cut ...]

INFO | ??:??:?? 83 [PowerFSM] Turn off screen

DEBUG | ??:??:?? 85 [Power] Battery: usbPower=1, isCharging=1, batMv=4250, batPct=100

DEBUG | ??:??:?? 105 [Power] Battery: usbPower=1, isCharging=1, batMv=4252, batPct=100

INFO | ??:??:?? 107 [Screen] Turn on screen

DEBUG | ??:??:?? 125 [Power] Battery: usbPower=1, isCharging=1, batMv=4253, batPct=100

INFO | ??:??:?? 127 [PowerFSM] Turn off screen

INFO | ??:??:?? 133 [Screen] Turn on screen

INFO | ??:??:?? 134 [Button] User toggled GpsMode. Now DISABLED

INFO | ??:??:?? 134 [Button] GPS power state move from ACTIVE to OFF

DEBUG | ??:??:?? 145 [Power] Battery: usbPower=1, isCharging=1, batMv=4256, batPct=100

Powered by [▲ Vercel](https://vercel.com/?utm_source=meshtastic&utm_campaign=oss) | Meshtastic® is a registered trademark of Meshtastic LLC. | [Legal Information](https://meshtastic.org/docs/legal).

Connected
```

lupusworax, 2024-12-07:
after updating to 2.5.15 or daily 2.5.16  startup is laggy and button presses are ignored, like it always is when gps is enabled, I just realized that once its done with the gps search and "gives up" I actually can enable and disable gps but it still cant find it anymore for some reason. 

```
All828
Info173
Debug691
Warn48
Error2
Disconnect

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 4 Filesystem files:

DEBUG | ??:??:?? 4  prefs (directory)

DEBUG | ??:??:?? 4  config.proto (183 Bytes)

DEBUG | ??:??:?? 4  module.proto (95 Bytes)

DEBUG | ??:??:?? 4  channels.proto (110 Bytes)

DEBUG | ??:??:?? 4  db.proto (8447 Bytes)

DEBUG | ??:??:?? 4  adafruit (directory)

DEBUG | ??:??:?? 4  bond_prph (directory)

DEBUG | ??:??:?? 4  bond_cntr (directory)

DEBUG | ??:??:?? 4 Power::lipoInit lipo sensor is not ready yet

DEBUG | ??:??:?? 4 Use analog input 31 for battery level

INFO | ??:??:?? 4 Scan for i2c devices

DEBUG | ??:??:?? 4 Scan for I2C devices on port 1

INFO | ??:??:?? 4 SSD1306 found at address 0x3c

INFO | ??:??:?? 4 SSD1306 found at address 0x3c

DEBUG | ??:??:?? 4 0x4 subtype probed in 2 tries

INFO | ??:??:?? 4 1 I2C devices found

DEBUG | ??:??:?? 4 acc_info = 0

INFO | ??:??:?? 4 S:B:63,2.5.15.79da2365

DEBUG | ??:??:?? 4 Reset reason: 0x0

DEBUG | ??:??:?? 4 Set random seed 3775063232

INFO | ??:??:?? 4 Init NodeDB

INFO | ??:??:?? 4 Load /prefs/db.proto

INFO | ??:??:?? 4 Loaded /prefs/db.proto successfully

INFO | ??:??:?? 4 Loaded saved devicestate version 23, with nodecount: 76

INFO | ??:??:?? 4 Load /prefs/config.proto

INFO | ??:??:?? 4 Loaded /prefs/config.proto successfully

INFO | ??:??:?? 4 Loaded saved config version 23

INFO | ??:??:?? 4 Load /prefs/module.proto

[... 1570 lines cut ...]
DEBUG | ??:??:?? 420 [GPS] Trying $PDTINFO (UM600)...

DEBUG | ??:??:?? 421 [GPS] Trying $PCAS06,1*1A (ATGM336H)...

DEBUG | ??:??:?? 421 [GPS] Trying $PCAS06,1*1A (ATGM332D)...

DEBUG | ??:??:?? 422 [GPS] Trying $PAIR021*39 (AG3335)...

DEBUG | ??:??:?? 422 [GPS] Trying $PAIR021*39 (AG3352)...

DEBUG | ??:??:?? 423 [GPS] Trying $PQTMVERNO*58 (LC86)...

DEBUG | ??:??:?? 423 [GPS] Trying $PCAS06,0*1B (L76K)...

DEBUG | ??:??:?? 424 [GPS] Trying $PMTK605*31 (L76B)...

DEBUG | ??:??:?? 424 [GPS] Trying $PMTK605*31 (PA1616S)...

WARN | ??:??:?? 426 [GPS] No GNSS Module (baudrate 777211975)

```

Powered by [▲ Vercel](https://vercel.com/?utm_source=meshtastic&utm_campaign=oss) | Meshtastic® is a registered trademark of Meshtastic LLC. | [Legal Information](https://meshtastic.org/docs/legal).

Connected

thebentern, 2024-12-07:
Something might have regressed as part of the changes in #5433 or #5447 as those are the only GPS related changes as a part of .15. Is your GPS module indeed an L76K or is it another mtk based module just being detected as that?

lupusworax, 2024-12-07:
its an ATGM336H

thebentern, 2024-12-07:
> its an ATGM336H

Excellent! I have a ton of these (and some RAK's already wired up to them). One of my favorites. I can try to reproduce this locally.

lupusworax, 2024-12-07:
> > its an ATGM336H
> 
> Excellent! I have a ton of these (and some RAK's already wired up to them). One of my favorites. I can try to reproduce this locally.

Same, they have  a super small footprint, work great  and are very inexpensive, I pretty much use them for all of my builds now ;)

fifieldt, 2024-12-07:
Tried using my closest combination - T114 (NRF52) and ATGM332D. Was not able to reproduce - detection and GPS worked fine.

```
DEBUG | ??:??:?? 26 [GPS] Probe for GPS at 9600

DEBUG | ??:??:?? 27 [GPS] Trying $PDTINFO (UC6580)...

DEBUG | ??:??:?? 27 [GPS] Trying $PDTINFO (UM600)...

DEBUG | ??:??:?? 28 [GPS] Trying $PCAS06,1*1A (ATGM336H)...

DEBUG | ??:??:?? 28 [GPS] $GPTXT,01,01,02,HW=ATGM332D,0000000000000*1A.

DEBUG | ??:??:?? 28 [GPS] Trying $PCAS06,1*1A (ATGM332D)...

DEBUG | ??:??:?? 28 [GPS] Found: $GPTXT,01,01,02,HW=ATGM332D

INFO | ??:??:?? 28 [GPS] ATGM332D detected, using GNSS_MODEL_ATGM336H Module
```

fifieldt, 2024-12-07:
Oh, interesting. I just noticed that your ATGM336H is detected as an L76K when it is "working", rather than an ATGM336H.

Can you make a firmware with GPS_DEBUG set to 1 in the variant.h and paste the log results here?

fifieldt, 2024-12-07:
Not a problem.

What I'm wondering is if your ATGM336H's on these new nodes reply slightly differently to other ATGM336Hs we've seen before. Easy to update once we get what it's saying.

Which variant are you using from the flasher? I can make a GPS_DEBUG build for you :)

lupusworax, 2024-12-08:
That would be great: 
[image]

thebentern, 2024-12-08:
Tested out fine for me on the latest with RAK4631 connected to ATGM336H module over UART. Detected on startup (as L76K which is normal) and got a GPS lock shortly thereafter.
```
DEBUG | ??:??:?? 1 [GPS] Probe for GPS at 9600
DEBUG | ??:??:?? 1 [GPS] Trying $PDTINFO (UC6580)...
DEBUG | ??:??:?? 2 [GPS] Trying $PDTINFO (UM600)...
DEBUG | ??:??:?? 2 [GPS] Trying $PCAS06,1*1A (ATGM336H)...
DEBUG | ??:??:?? 3 [GPS] Trying $PCAS06,1*1A (ATGM332D)...
DEBUG | ??:??:?? 3 [GPS] Trying $PAIR021*39 (AG3335)...
DEBUG | ??:??:?? 4 [GPS] Trying $PAIR021*39 (AG3352)...
DEBUG | ??:??:?? 4 [GPS] Trying $PQTMVERNO*58 (LC86)...
DEBUG | ??:??:?? 5 [GPS] Trying $PCAS06,0*1B (L76K)...
INFO  | ??:??:?? 5 [GPS] L76K detected, using GNSS_MODEL_MTK Module
DEBUG | ??:??:?? 5 [GPS] Publish pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | ??:??:?? 5 [GPS] No GPS lock
DEBUG | ??:??:?? 5 [GPS] onGPSChanged() pos@0 time=0 lat=0 lon=0 alt=0
INFO  | ??:??:?? 5 [GPS] updatePosition LOCAL pos@0 time=0 lat=0 lon=0 alt=0
DEBUG | ??:??:?? 5 [GPS] Set local position: lat=0 lon=0 time=0 timestamp=0
```

lupusworax, 2024-12-08:
could the Promicro NRF52 dev board could be the "culprint"?  I am not owning a RAK so no clue how it would behave there. I am having this problem on a FakeTec build. 
https://github.com/gargomoma/fakeTec_pcb

lupusworax, 2024-12-08:
cheese I feel stupid right now, there is a new bootloader available for the Pro micro NRF52 dev board. 0.9.2 after updating it and flashing the most recent nightly build it works normal again........Sorry I had no clue that a bootloader is so critical for this to work?!

lupusworax, 2024-12-08:
4 boards now all working after updating to newest bootloader version.  I tested all of them by just updating to recent firmware, all of them could not find gps after the update all work fine now.

closed 2024-12-08 by thebentern

jack-changyan, 2025-01-05:
@lupusworax  
What version of bootloader and firmware do you use? Thank you very much that I also encountered a similar problem, which has been bothering me for several days.

lupusworax, 2025-01-05:
I updated to 9.0.2, then flashed an older version and then the recent one and it started working again. But just today I had the same issue again with a node that was already on 0.9.2  I then flashed an older firmware again and then a the most recent and it started working again AFTER the initial bootup was kind of buggy.......weird but it works now, give it a few minutes.......its honestly a bit weird.

jack-changyan, 2025-01-06:
> I updated to 9.0.2, then flashed an older version and then the recent one and it started working again. But just today I had the same issue again with a node that was already on 0.9.2 I then flashed [...]
@lupusworax 
I'm using the latest nice_nano_bootloader-0.9.2_s140_6.1.1.zip and this version of the firmware nrf52_promicro_diy_xtal-v2.5.19.6aabbedc.daily.uf2, My GPS module is E108-GN03, what is your GPS model? :)

lupusworax, 2025-01-06:
I am using atgm336h pretty much all the time as it is, tiny, inexpensive and works great!

jack-changyan, 2025-01-06:
@lupusworax  
The compatibility of the ATGM336H module is notably robust.  I am very interested in learning more about the power consumption characteristics of this module during its operation.  the price in China is approximately 16 RMB, which is roughly equivalent to two US dollars?  Thank you very much for your assistance.