meshtastic/firmware #5524: [Bug]: Lilygo T-Deck backlight toggle starts message.

bmmcwhirt, 2024-12-07:
### Category

Other

### Hardware

T-Deck

### Firmware Version

2.5.14.f2ee0df Beta

### Description

Hardware: Lilygo T-Deck Plus 915Mhz White (H737-01)

The [alt+b] shortcut to enable keyboard backlight does toggle the backlight, but it also begins a network message input with the character 'b'.

This behavior is not present with the [alt c} shortcuts.

Process to reproduce:
    1. Press and hold 'alt'
    2. Press and release 'b'
    3. Release 'alt'
 
Expected behavior: 
    Backlighting should toggle and no message should be started.

Actual behavior:
    Backlighting is toggled and a new message is started with the character 'b'.

I do not currently have logs or a sanitized dump of my config. I can provide those when I get a moment to connect this to my dev machine.

joyel24, 2024-12-11:
Lilygo sent me a Tdeck+ which is coming very soon.
I'll try to see if I can fix it...

joyel24, 2024-12-15:
@bmmcwhirt I investigated a bit on it...
Keyboard backlight is not handled by the firmware but by the MCU inside the keyboard itself so it's not going to be an easy fix as I expected before seeing how it's done.

joyel24, 2024-12-17:
The fix is coming to the keyboard firmware but every affected tdeck will need to be physically plugged to the esp32-c3 pins to upgrade it.

closed 2024-12-17 by thebentern

joyel24, 2024-12-18:
https://github.com/Xinyuan-LilyGO/T-Deck/issues/69#issuecomment-2550053922

Fixed and will be flashed in the factory for the future devices. But as I said, to fix that you'll need to manually flash the esp-c3 controlling the keyboard.