meshtastic/firmware #5526: [Bug]: Mixture of SNR and LAT/LONG

wdrei, 2024-12-07:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.5.15.79da236

### Description

I have carried out a range test. I noticed that there are occasional jumps in the position and at the same time in the signal-to-noise ratio.

The most striking cases are those where the position jumps, then the SNR changes and finally a jump back to the original position appears in the log.

The information on the position and the signal-to-noise ratio is probably not always read out correctly.

I suspect that there could be a sign problem.

The resulting range test file is as attached here: 
[rangetest.csv](https://github.com/user-attachments/files/18049999/rangetest.csv)

The conspicuous entries are between lines 50 and 60. There, the position changes immediately, then remains the same, then returns and remains the same again.

fifieldt, 2024-12-08:
Thanks for sending this in, we appreciate logs :)

May I confirm the hardware of the sender and the receiver?

fifieldt, 2024-12-21:
I spent some time reading the code.

For information: the RX Lat and RX Lon is de-coupled from the RX_SNR. What happens is the RangeTestModule processes an incoming packet. That packet has an SNR from the receiving radio, which is what is stored in the CSV. At the same time as its storing that information, the RangeTestModule looks to GPSStatus to get the latest lat/lon/altitude for the node on which it is running. That information could be from a different time to the receipt of the packet, based on the settings for the GPS. I am guessing that your GPS is set to run reasonably frequently, so it should line up fairly well?

Here's a theory -- during the times the SNR goes bad in the logs, the GPS is running and causing some kind of hardware issue that makes the reception worse? Does that seem possible? To test -- change the frequency at which the GPS runs to a bit longer and see if the issue appears along those longer timelines?

fifieldt, 2024-12-21:
As this requires more troubleshooting before we can confirm a bug, transferring to a discussion.