meshtastic/firmware #5549: [Bug]: Memory leaks in MQTT::onReceive

esev, 2024-12-10:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.5.16.f81d3b0

### Description

I believe there are two potential memory leaks in `src/mqtt/MQTT.cpp` `MQTT::onReceive`

1. https://github.com/meshtastic/firmware/blob/761a99d2411093f8902d23e513c13ffc429b49e2/src/mqtt/MQTT.cpp#L148 Any pointers within the `meshtastic_ServiceEnvelope` that are non-NULL are leaked.

2. https://github.com/meshtastic/firmware/blob/761a99d2411093f8902d23e513c13ffc429b49e2/src/mqtt/MQTT.cpp#L202 The `meshtastic_MeshPacket *p` is leaked when the `if` statement evaluates to `false`

There are several places in MQTT::onReceive where cleanup is needed. If I may offer a suggestion; use std::unique_ptr to help ensure these are always freed/released on `return`

Example:
```c++
// make sure to free both strings and the MeshPacket (passing in NULL is acceptable)
struct ServiceEnvelopeCleaner {
    void operator()(meshtastic_ServiceEnvelope *e)
    {
        free(e->channel_id);
        free(e->gateway_id);
        free(e->packet);
    }
};
using ScopedServiceEnvelopeReleaser = std::unique_ptr<meshtastic_ServiceEnvelope, ServiceEnvelopeCleaner>;
struct PacketPoolMeshPacketDeleter {
    void operator()(meshtastic_MeshPacket *p)
    {
        packetPool.release(p);
    }
};
using UniqueMeshPacket = std::unique_ptr<meshtastic_MeshPacket, PacketPoolMeshPacketDeleter>;

...

ScopedServiceEnvelopeReleaser free_e(&e);

...

UniqueMeshPacket p(packetPool.allocCopy(*e.packet));

...

router->enqueueReceivedMessage(p.release());
```

esev, 2024-12-10:
> Great analysis! Would you be willing to submit a PR making the proposed changes?

I would love to, and normally I would. But I can't currently sign the CLA due to [restrictions at work](https://opensource.google/documentation/reference/patching#no-review). I'm working on resolving any conflicts there, but can't send PRs for now.

Matzebhv, 2024-12-12:
Same as here -> https://github.com/meshtastic/firmware/issues/5458

leshniak, 2024-12-18:
Thank you @esev for the contribution, I'm already testing your changes on my device.

Matzebhv, 2024-12-18:
I will test as soon as this pull is available in the coming alpha.

PR #5592 (2024-12-19): Refactor MQTT::onReceive to reduce if/else nesting
Refactoring using std::unique_ptr and variable scoping to help improve readability and reduce the chance for memory leaks.

fixes #5549
fixes #5458
fixes #5559