meshtastic/firmware #5559: [Bug]: WiFi disconnects after some time when using MQTT

chairmanfaust, 2024-12-13:
### Category

WiFi

### Hardware

Station G2

### Firmware Version

2.5.14

### Description

I have a station G2 acting as an MQTT gateway via wifi. After about a day or two i can no longer use the app to connect via wifi address. My router shows that the device is connected and still has an IP address but it no longer responds to ping. The behavior is identical to a closed bug found here: https://github.com/meshtastic/firmware/issues/4725

Matzebhv, 2024-12-13:
Same problem here with 2 t-beam supreme. You can try to update to > .14 as it seems that this fixed the problem for some people. For me and others this does not work but is worth the try.
You can also look here -> https://github.com/meshtastic/firmware/issues/5549 and here -> https://github.com/meshtastic/firmware/issues/5458
For me it seems to be an ESP related problem.

koliha, 2024-12-16:
Spoke too soon in #5458 - I managed 10 days 17hrs of uptime before it exhausted all heap memory and stopped responding.  Looks like a memory leak, probably as outlined in #5549 

Ping was still semi-responsive.  Hitting the http/json page was impossible.  It also dropped off MQTT.

[image]
[image]

esev, 2024-12-17:
Can you confirm that you are **not** using JSON mode for MQTT and have downlink enabled? If that's correct, I suspect this due to the memory leaks from #5549. If that's not correct, then there is another issue to troubleshoot.

chairmanfaust, 2024-12-17:
I am using JSON on my mqtt server and have downlink enabled on my primary (longfast) channel as well as 2 other channels 

[image]

[image]

esev, 2024-12-17:
My mistake. I hadn't realized that [MQTT::send subscriptions()](https://github.com/meshtastic/firmware/blob/4edeca5f846024365bbd8855a454d42d2eae4d57/src/mqtt/MQTT.cpp#L409) subscribes to both JSON and protobuf channels. It's probably the same memory leaks from the previously mentioned issue.

garthvh, 2024-12-17:
JSON functionality in the firmware is in the process of being retired https://github.com/meshtastic/firmware/issues/5507

esev, 2024-12-17:
@garthvh The memory leak is not in the JSON code-path. #5549 for details; essentially the same bug.

EnGamma, 2024-12-18:
I'm seeing similar/same symptoms of losing phone-node wifi connection with TLORA_V2_1_1P6 device fw 2.5.11.   I'm sometimes able to reboot the node from another using RF remote administration.   I have 4 identical HW nodes all on fw 2.5.11, but it is affecting 2 nodes more than the others and one node especially.

These symptoms started after fw update from 2.3.10 to 2.5.11.

PR #5592 (2024-12-19): Refactor MQTT::onReceive to reduce if/else nesting
Refactoring using std::unique_ptr and variable scoping to help improve readability and reduce the chance for memory leaks.

fixes #5549
fixes #5458
fixes #5559
closed 2024-12-19 by thebentern