meshtastic/firmware #5560: [Bug]: DFrobot Lark Sensor No data to be seen via I2C

MALAONE1, 2024-12-13:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

2.5.16

### Description

I have a DFrobot Lark sensor connected to I2C with a RAK 4631 with RAK 19003. The Lark writes all weather data to the internal memory. I2C communication is set in the config, the green LED lights up on the Lark which means that I2C is activated, all cables have been checked several times for the I2C interface to the RAK. According to the serial monitor, the LARK is not recognized.

caveman99, 2024-12-14:
do you have a bootup log where the i2c scanner is running?

MALAONE1, 2024-12-14:
> do you have a bootup log where the i2c scanner is running?

The Lark is not visible in the serial monitor. The only thing that was detected on 0x40 is the also installed INA219. I also tried to uninstall this and only install the Lark. However, it is not displayed there. I will submit a copy of the output during the boot.

MALAONE1, 2024-12-15:
> do you have a bootup log where the i2c scanner is running?

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 3 Filesystem files:

DEBUG | ??:??:?? 3  RAK (108 Bytes)

DEBUG | ??:??:?? 3  adafruit (directory)

DEBUG | ??:??:?? 3  bond_prph (directory)

DEBUG | ??:??:?? 3  bond_cntr (directory)

DEBUG | ??:??:?? 3  MODE (1 Bytes)

DEBUG | ??:??:?? 3  prefs (directory)

DEBUG | ??:??:?? 3  channels.proto (160 Bytes)

DEBUG | ??:??:?? 3  module.proto (115 Bytes)

DEBUG | ??:??:?? 3  config.proto (166 Bytes)

DEBUG | ??:??:?? 3  db.proto (4978 Bytes)

DEBUG | ??:??:?? 3 Power::lipoInit lipo sensor is not ready yet

DEBUG | ??:??:?? 3 Use analog input 5 for battery level

INFO | ??:??:?? 3 Scan for i2c devices

DEBUG | ??:??:?? 3 Scan for I2C devices on port 1

DEBUG | ??:??:?? 3 Register MFG_UID: 0x2000

INFO | ??:??:?? 3 INA219 found at address 0x40

DEBUG | ??:??:?? 3 Register MFG_UID FE: 0xffff

DEBUG | ??:??:?? 3 Register MFG_UID 05: 0xffff

INFO | ??:??:?? 3 1 I2C devices found

DEBUG | ??:??:?? 3 acc_info = 0

INFO | ??:??:?? 3 S:B:9,2.5.16.f81d3b0

INFO | ??:??:?? 3 Build timestamp: 1733662250

DEBUG | ??:??:?? 3 Reset reason: 0x0

DEBUG | ??:??:?? 4 Set random seed 70880291

INFO | ??:??:?? 4 Init NodeDB

INFO | ??:??:?? 4 Load /prefs/db.proto

INFO | ??:??:?? 4 Loaded /prefs/db.proto successfully

INFO | ??:??:?? 4 Loaded saved devicestate version 23, with nodecount: 56

INFO | ??:??:?? 4 Load /prefs/config.proto

INFO | ??:??:?? 4 Loaded /prefs/config.proto successfully

INFO | ??:??:?? 4 Loaded saved config version 22

INFO | ??:??:?? 4 Load /prefs/module.proto

INFO | ??:??:?? 4 Loaded /prefs/module.proto successfully

INFO | ??:??:?? 4 Loaded saved moduleConfig version 23

INFO | ??:??:?? 4 Load /prefs/channels.proto

INFO | ??:??:?? 4 Loaded /prefs/channels.proto successfully

INFO | ??:??:?? 4 Loaded saved channelFile version 23

DEBUG | ??:??:?? 4 cleanupMeshDB purged 4 entries

DEBUG | ??:??:?? 4 Use nodenum 0x2923a3b3

DEBUG | ??:??:?? 4 Expand short PSK #1

INFO | ??:??:?? 4 Wanted region 3, using EU_868

DEBUG | ??:??:?? 4 Coerce telemetry to min of 30 minutes on defaults

DEBUG | ??:??:?? 4 Use GPIO09 for button

DEBUG | ??:??:?? 4 Use compiled/slipstreamed tzplaceholder

DEBUG | ??:??:?? 4 Set Timezone to GMT0

DEBUG | ??:??:?? 4 NeighborInfoModule is disabled

DEBUG | ??:??:?? 4 Rescan for I2C keyboard

DEBUG | ??:??:?? 4 Scan for I2C devices on port 1

DEBUG | ??:??:?? 4 Scan address 0x1f

DEBUG | ??:??:?? 4 Scan address 0x55

DEBUG | ??:??:?? 4 Scan address 0x5a

DEBUG | ??:??:?? 4 Scan address 0x5f

ERROR | ??:??:?? 4 Could not open / read /prefs/ringtone.proto

INFO | ??:??:?? 4 Init External Notification Module

INFO | ??:??:?? 4 Use Pin 36 in digital mode

DEBUG | ??:??:?? 4 SX126xInterface(cs=42, irq=47, rst=38, busy=46)

DEBUG | ??:??:?? 4 SX126X_DIO3_TCXO_VOLTAGE defined, using DIO3 as TCXO reference voltage at 1.800000 V

INFO | ??:??:?? 4 Start meshradio init

INFO | ??:??:?? 4 Radio freq=869.525, config.lora.frequency_offset=0.000

INFO | ??:??:?? 4 Set radio: region=EU_868, name=LongFast, config=0, ch=0, power=27

INFO | ??:??:?? 4 myRegion->freqStart -> myRegion->freqEnd: 869.400024 -> 869.650024 (0.250000 MHz)

INFO | ??:??:?? 4 numChannels: 1 x 250.000kHz

INFO | ??:??:?? 4 channel_num: 1

INFO | ??:??:?? 4 frequency: 869.525024

INFO | ??:??:?? 4 Slot time: 77 msec

INFO | ??:??:?? 4 Set radio: final power level=22

INFO | ??:??:?? 4 SX126x init result 0

INFO | ??:??:?? 4 Frequency set to 869.525024

INFO | ??:??:?? 4 Bandwidth set to 250.000000

INFO | ??:??:?? 4 Power output set to 22

MALAONE1, 2024-12-17:
Yes, there seems to be a problem. Is there any information about it yet?

caveman99, 2024-12-19:
i changed the detection code to not mistake it for the RAK GPS. I don't have a lark myself, i can test with @MALAONE1 directly.

caveman99, 2024-12-19:
umm, you have an INA219 at the same board? that won't work. we only support one device per I2C id.

windbender, 2025-01-08:
I've been trying to get a Lark to source data into a T-Beam.  It doesn't appear to work with 2.5.15.    I'd be up for using an Alpha version if that's something I could test. (I had previously had a BME280, which was sending data ).

fifieldt, 2025-01-09:
@windbender , are you able to share logs of the early phases? From that I'll try to get the I2C address and ship you a build for testing ...

windbender, 2025-01-09:
@fifieldt 
I'm hoping the following is what you're looking for. I will note that the I2C stuff looks quite like that of MALAONE1

```
rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (858) esp_core_dump_flash: No core dump??ѥѥ???found!
E (858) esp_core_dump_flash: No core dump partition found!
[    11][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   474][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
??@INFO  | ??:??:?? 0 

//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 0 Filesystem files (905216/1048576 Bytes):
DEBUG | ??:??:?? 0  /prefs/bsec.dat (221 Bytes)
DEBUG | ??:??:?? 0  /prefs/channels.proto (102 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (216 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (448 Bytes)
DEBUG | ??:??:?? 0  /prefs/module.proto (106 Bytes)
DEBUG | ??:??:?? 0  /static/.gitkeep (0 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_Black.svg.gz (602 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_White.svg.gz (610 Bytes)
DEBUG | ??:??:?? 0  /static/apple-touch-icon.png.gz (3164 Bytes)
DEBUG | ??:??:?? 0  /static/favicon.ico.gz (2270 Bytes)
DEBUG | ??:??:?? 0  /static/icon.svg.gz (852 Bytes)
DEBUG | ??:??:?? 0  /static/index-b9vwykds.js.gz (222120 Bytes)
DEBUG | ??:??:?? 0  /static/index-wV0rbIjU.css.gz (16573 Bytes)
DEBUG | ??:??:?? 0  /static/index.html.gz (547 Bytes)
DEBUG | ??:??:?? 0  /static/mapbox-gl-DY02hNYD.js.gz (391221 Bytes)
DEBUG | ??:??:?? 0  /static/maplibre-gl-H6mDd47p.js.gz (210429 Bytes)
DEBUG | ??:??:?? 0  /static/robots.txt.gz (42 Bytes)
DEBUG | ??:??:?? 1  /static/site.webmanifest.gz (197 Bytes)
[  1021][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
[  1036][I][XPowersCommon.tpp:445] begin(): SDA:21 SCL:22
[  1041][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
WARN  | ??:??:?? 1 No AXP2101 power management
[  1050][I][XPowersAXP2101.tpp:216] ~XPowersAXP2101(): ~XPowersAXP2101
[  1061][I][XPowersCommon.tpp:445] begin(): SDA:21 SCL:22
[  1066][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
INFO  | ??:??:?? 1 AXP192 PMU init succeeded
[  1082][D][XPowersAXP192.tpp:2026] setInterruptImpl(): setInterruptImpl DISABLE - 0xffffffffff

DEBUG | ??:??:?? 1 DC1  : +   Voltage:3300 mV 
DEBUG | ??:??:?? 1 DC2  : -   Voltage:1800 mV 
DEBUG | ??:??:?? 1 DC3  : +   Voltage:3300 mV 
DEBUG | ??:??:?? 1 LDO2 : +   Voltage:3300 mV 
DEBUG | ??:??:?? 1 LDO3 : +   Voltage:3300 mV 
DEBUG | ??:??:?? 1 Battery: usbPower=1, isCharging=1, batMv=3341, batPct=6
INFO  | ??:??:?? 1 Scan for i2c devices
[  1145][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
DEBUG | ??:??:?? 1 Scan for I2C devices on port 1
INFO  | ??:??:?? 1 AXP192/AXP2101 found at address 0x34
INFO  | ??:??:?? 1 SSD1306 found at address 0x3c
INFO  | ??:??:?? 1 SSD1306 found at address 0x3c
DEBUG | ??:??:?? 1 0x4 subtype probed in 2 tries 
INFO  | ??:??:?? 1 2 I2C devices found
DEBUG | ??:??:?? 1 acc_info = 0
INFO  | ??:??:?? 1 S:B:4,2.5.15.79da236
INFO  | ??:??:?? 1 Build timestamp: 1732889288
DEBUG | ??:??:?? 1 Total heap: 190528
DEBUG | ??:??:?? 1 Free heap: 156324
DEBUG | ??:??:?? 1 Total PSRAM: 4192107
[... 88 lines cut ...]
DEBUG | ??:??:?? 9 [GPS] No GPS lock
DEBUG | ??:??:?? 9 [GPS] onGPSChanged() pos@0 time=0 lat=0 lon=0 alt=0
INFO  | ??:??:?? 9 [GPS] updatePosition LOCAL pos@0 time=0 lat=0 lon=0 alt=0
DEBUG | ??:??:?? 9 [GPS] Set local position: lat=0 lon=0 time=0 timestamp=0
DEBUG | ??:??:?? 9 [GPS] Node status update: 1 online, 1 total
WARN  | ??:??:?? 9 [DetectionSensor] Detection Sensor Module: Set to enabled but no monitor pin ie
INFO  | ??:??:?? 9 [RangeTest] Range Test Module - Disabled
INFO  | ??:??:?? 9 [PowerFSM] Init the NimBLE bluetooth module
INFO  | ??:??:?? 10 [Screen] Done with boot screen
DEBUG | ??:??:?? 10 [Screen] Show standard frames
DEBUG | ??:??:?? 10 [Screen] EnvironmentTelemetry wants a UI Frame
DEBUG | ??:??:?? 10 [Screen] Show 1 module frames
DEBUG | ??:??:?? 10 [Screen] Total frame count: 104
DEBUG | ??:??:?? 10 [Screen] Added modules.  numframes: 1
DEBUG | ??:??:?? 10 [Screen] Finished build frames. numframes: 3
DEBUG | ??:??:?? 10 [GPS] NMEA GPS time 2025-01-09 06:08:08 age 3
DEBUG | ??:??:?? 10 [GPS] Upgrade time to quality GPS
DEBUG | 22:08:08 10 [GPS] Read RTC time as 1736402888
DEBUG | 22:08:08 10 [GPS] Publish pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 22:08:08 10 [GPS] onGPSChanged() pos@0 time=1736402888 lat=0 lon=0 alt=0
INFO  | 22:08:08 10 [GPS] updatePosition LOCAL pos@0 time=1736402888 lat=0 lon=0 alt=0
DEBUG | 22:08:08 10 [GPS] Set local position: lat=0 lon=0 time=1736402888 timestamp=0
DEBUG | 22:08:08 10 [GPS] Node status update: 1 online, 1 total
INFO  | 22:08:09 11 [EnvironmentTelemetry] Environment Telemetry: init
```

fifieldt, 2025-01-09:
Thanks. So from what I understand, the lark should appear at 0x42 ...

In the logs from @MALAONE1 , I can see it is seeing something at 0x42
```
DEBUG | ??:??:?? 3 Register MFG_UID FE: 0xffff
DEBUG | ??:??:?? 3 Register MFG_UID 05: 0xffff
```

but it doesn't match because we expect 0x5305 and get 0xffff

However, in the logs from @windbender , I don't see anything coming up at 0x42 .

fifieldt, 2025-01-09:
I've emailed DFRobot to ask the expected return values for the ID register.

windbender, 2025-01-09:
The code examples on this page DO indicated 0x42 as the I2c address:
https://wiki.dfrobot.com/SKU_EDU0157_EN_Lark_Weather_Station

closed 2025-03-19 by github-actions[bot]