meshtastic/firmware #5566: [Feature Request]: Define BUTTON_PIN for rp2040-lora

enif1, 2024-12-14:
### Platform

RP2040

### Description

After easily adding an OLED display to my rp2040-lora, I thought adding a button would be just as easy.  But all my trials were without success. When I finally looked at the source, I realized that for this device BUTTON_PIN was undefined, since the usual pin 17 was already used for LORA_DIO4.  Looking  further into the source, I realized that for a pin to be assigned to the button in the device configuration, BUTTON_PIN must be defined in the source.

Thus my feature request: Would it be possible to change rp2040-lora/variant.h to (re)define BUTTON_PIN to some unused pin (e.g. pin 21) instead of undefining it?  This way, the corresponding code would be compiled into the firmware and the user could easily add a button to any available pin (redefining the pin if necessary in the device configuration).

Thank you!!!

fifieldt, 2024-12-14:
In the app you can define PIN_BUTTON under device settings.

enif1, 2024-12-14:
@fifieldt: If I understand the source code correctly, what you can define in device settings is config.device.button_gpio. But BUTTON_PIN is a macro (which cannot be modified a run time) and all the program part dealing with the button is only compiled if BUTTON_PIN is defined (#ifdef BUTTON_PIN).  So BUTTON_PIN does need to be defined at the source level for the button to actually work.

fifieldt, 2024-12-14:
Apologies, you are correct. For any variant that doesn't originally have a BUTTON_PIN defined, that app setting won't do anything.

fifieldt, 2024-12-15:
Had some chats in discourse with other developers and we're going to try a general fix to this so you'll be able to use the app to define the pin. However, I'll also try putting a patch specific to the RP2040-lora shortly.

enif1, 2024-12-15:
Thank you -  that's great!!!

closed 2024-12-15 by thebentern