meshtastic/firmware #5568: 🐞 [Bug]: Exchange Positions sends Position message with precision_bits set to 0

oherrala, 2024-12-14:
### Firmware Version

2.5.12

### What did you do?

Go to Nodes list, select node, scroll down and press "Exchange Positions" button.

I monitor what my node sends via separate node sending traffic into MQTT server.

### Expected Behavior

When pressing exchange position button I'd expect it to send position report into channel with precision that is set in channel settings. At least it should not send it with precision_bits set to 0 since it is documented as "A value of 0 means that location data is never sent over the given channel."[^1]

[^1]: https://meshtastic.org/docs/configuration/radio/channels/#position-precision

### Current Behavior

Currently exchange position button sends the precise GPS location with precision_bits set to 0. This was surprising because it doesn't obey the channel settings of precision.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GUVWAF, 2024-12-14:
The firmware doesn't do this for direct messages, as we discussed in #3752.

oherrala, 2025-02-20:
How should we interpret this precision_bits value of 0? Now the functionality and documentation have a mismatch. Is 0 precision same as 32?

Currently this causes seems to be a undocumented special cases for handling position reports. For example I filter out position reports with precision less than 16 meaning these 0 precision bit values are left out as well.

closed 2025-04-15 by github-actions[bot]