meshtastic/firmware #5569: [Feature Request]: Detection sensor accumulate count between reports

seagull9000, 2024-12-14:
### Platform

other

### Description

Some sensors may trigger a burst of activity that would not be friendly to the mesh.  For example, a rain sensor may trigger rapidly during a torrential downpour  The minimum reporting interval would mitigate this impact, but with the current functionality, valuable information would be lost.

Proposed enhancement:
1.  Should have: Accumulate a count of sensor activations between reports.  This could be a lifetime count (not reset after each report) so that missed reports will not compromise the integrity of the data (client application can just subtract current received from previous received).
2.  Nice to have: Attribute a quantitative value for each activation.  For example, for a rain gauge, the amount in millilitres per activation could be entered and accummulated as in 1.

The CLI/API could have a function to reset accummulated values to 0.  This would provide flexibility so that client applications don't have to worry about starting values.

garthvh, 2024-12-14:
There is already an interval for how frequently the message is sent out, this is how you reduce mesh traffic, it also no longer works on the default channel if you upgrade your firmware.

seagull9000, 2024-12-14:
> There is already an interval for how frequently the message is sent out, this is how you reduce mesh traffic, it also no longer works on the default channel if you upgrade your firmware.

Yes, thanks, but the issue is with the loss of data if multiple detections occur between the minimum interval.  I tried to find the relevant protobuf to see if a count already exists, but not sure which it is.

The key point is that messages can be missed, so if the count is important, we can't rely on the client application to keep track of it if the message is lost.