meshtastic/firmware #5587: [Bug]: Don't allow message sending from nodeid of 4294967295

garthvh, 2024-12-15:
### Category

BLE, Serial, WiFi

### Hardware

Linux Native, DIY, Other

### Firmware Version

All

### Description

After trapping some additional cases in the iOS app that were generating ? for the messaging user there is at least one user on the public MQTT for the US that has a node number of 4294967295 (the uint32 max value used to indicate a message is a channel message and not a DM)

We should prevent messages from being sent from the magic number.

A client notification could be sent if someone attempts to send a message from this nodenum.

GUVWAF, 2024-12-15:
This is likely not a real node, but just someone crafting a packet on MQTT. 

We already don't allow setting the `from`: 
https://github.com/meshtastic/firmware/blob/09c082fd004423374f1decd880ee7a448d6e3999/src/mesh/MeshService.cpp#L176

And you can't get this NodeNum. 

Also, newer firmware will ignore incoming messages from the broadcast NodeNum: 
https://github.com/meshtastic/firmware/blob/09c082fd004423374f1decd880ee7a448d6e3999/src/mesh/Router.cpp#L664-L665

thebentern, 2024-12-15:
I hesitate the make this a clientnotification as it could provide incentive for bad actions to create annoying push notifications. As @GUVWAF mentioned we already handle these cases in the firmware, but there is nothing preventing someone from pushing the data into the broker by forging a packet with a python script for instance. We can probably block this in the broker though.

thebentern, 2024-12-16:
For posterity, this issue was fixed at the Router level in the v2.5.6 release of the firmware

closed 2024-12-16 by thebentern