meshtastic/firmware #5596: [Bug]: case sensitive ESSID and Bluetooth lockout

mcccol, 2024-12-17:
### Category

WiFi

### Hardware

T-Beam

### Firmware Version

2.5.15.79da236

### Description

I accidentally/carelessly typed a lower case letter into the SSID field in network configuration over bluetooth. Because meshtastic uses an old version of esp32 libraries, (because, I presume, meshtastic uses arduino ide and it's very hard to stay up-to-date when you make that choice,) bluetooth becomes unavailable when wifi is being attempted. Wifi connection will, in this case, always fail. You cannot use the serial console to correct the error, as (it seems to me) the serial console stores SSID lower cased. So, the whole thing is livelocked. It's also quite hard to find and fix.

In my opinion, the best first move (apart from fixing the serial interface) would be to *caselessly* compare the scanned SSIDs returned by esp32 library scanner, and use the (likely) one which matches, if only in the case that there are otherwise no matches.

The long-run fix is to upgrade to more recent esp32 libs. I understand that this would be a lot of effort. I, personally, would (and do) use espidf for such large scale developments as meshtastic.

GUVWAF, 2024-12-17:
> I presume, meshtastic uses arduino ide and it's very hard to stay up-to-date when you make that choice,

No, Meshtastic uses PlatformIO. 

> The long-run fix is to upgrade to more recent esp32 libs.  

The current version of the platform we’re using for ESP32 is based on [ESP-IDF](https://docs.platformio.org/en/latest/frameworks/espidf.html) v5.3.1, which is very recent. Enabling Wi-Fi completely disables Bluetooth and this is deliberately, because the coexistence provided by ESP32 is a simple [time-slice](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/coexist.html#coexistence-period-and-time-slice) based mechanism and it will always influence the performance of either method.

> In my opinion, the best first move (apart from fixing the serial interface) would be to caselessly compare the scanned SSIDs returned by esp32 library scanner, and use the (likely) one which matches, if only in the case that there are otherwise no matches.

To be honest, this sounds like a lot of work for fixing a user error. I’m not sure which “serial interface” you are referring to, but it should be possible to recover from this using the Python CLI.

> personally, would (and do) use espidf for such large scale developments as meshtastic. 

ESP-IDF is not an option with the same codebase if you support multiple targets like ESP32, nRF52, RP2xxx, STM32 and Linux. However, the Arduino core is based on ESP-IDF.

closed 2024-12-17 by thebentern

garthvh, 2024-12-17:
SSID's are indeed case sensitive

mcccol, 2024-12-17:
Thank you for correcting my incorrect presumptions, and clearly explaining your design and project choices where I have raised issues.

> The current version of the platform we’re using for ESP32 is based on ESP-IDF v5.3.1, which is very recent. Enabling Wi-Fi completely disables Bluetooth and this is deliberately, because the [...]

That is sound. Might I suggest that the period between acquisition of a wifi connection (say, up to the point of dhcp completing) would be a place to moderate that (otherwise sound) tradeoff?
 
I see that the project is optimised to 'just work' when it does (when users haven't made stupid errors, as I had.) In that configured state, bluetooth is superfluous. In the process of configuring the coexistence of bluetooth would have been useful once I had used the serial console, diagnosed what error 201 from the library might mean, and was trying to change a lower to an upper-case "O" in the SSID.

To that end, and since the system remains configurable under python CLI the whole time, would it be worth relying upon the user setting for bluetooth to make the decision to optimise/deoptimise the wifi/bluetooth choice? At least until wifi is actually launched on the device.

> To be honest, this sounds like a lot of work for fixing a user error. I’m not sure which “serial interface” you are referring to, but it should be possible to recover from this using the Python CLI.

It is a lot of work, and a lot of code to pack into a little embedded system, for very little benefit. For what it's worth, the serial interface to which I referred was https://client.meshtastic.org/ (with which I struggled to find, and correct, my error.) I will find, install, and use the python cli in future for such cases.
 
> > personally, would (and do) use espidf for such large scale developments as meshtastic.
> 
> ESP-IDF is not an option with the same codebase if you support multiple targets like ESP32, nRF52, RP2xxx, STM32 and Linux. However, the Arduino core is based on ESP-IDF.

Thanks for your feedback, and the benefit of your work.