meshtastic/firmware #5612: [Feature Request]: Please put OK to MQTT for LoRa to MQTT settings

Dtrieb, 2024-12-19:
### Contact Details

dtrieb

### Tell us your idea.

People often forget to enable "Okay to MQTT" because it's under LoRa settings.

Please add this both to MQTT settings.

- **Okay to MQTT**
- **Ignore MQTT**

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

garthvh, 2024-12-19:
All of the config settings are protobufs messages shared across platforms, this setting is on Lora config in all of the clients because it is a part of that protobuf message

GUVWAF, 2024-12-19:
It's in the LoRa settings because it controls whether someone else can uplink your packet when they receive it over LoRa.
The firmware itself doesn't check the "OK to MQTT" bit for your *own* packets, but the public MQTT server will still drop them if it's not set. 

Maybe it makes sense to automatically enable it when enabling MQTT?

garthvh, 2024-12-19:
I think users need to continue to explicitly choose this setting, especially while we don't yet have clarity on the PII aspects or a clear policy on the use of mqtt data by third parties.

GUVWAF, 2024-12-19:
OK, then it makes even more sense to me this is in the LoRa settings, which you definitely need to go into to set your region in the first place. A user that doesn't use MQTT still has to decide whether others can use their data. Same for "ignore MQTT", even if you don't use it, you want to set whether you want to receive packets that have been routed via MQTT somewhere.

closed 2024-12-19 by thebentern