meshtastic/firmware #5614: [Bug]: T114 GPS won't update

aarondong2005, 2024-12-19:
### Category

Other

### Hardware

Heltec Mesh Node T114

### Firmware Version

2.5.15

### Description

I have Heltec V3 and T114. With same setting the V3 will update without issue for the route. But the T114 only update once it get the signal and no update at all

fifieldt, 2024-12-20:
Hi, could you define "won't update" in a bit more detail?

Also, is it possible to provide log files, such as those from the serial console at flasher.meshtastic.org

aarondong2005, 2024-12-20:
I am so sorry. I mean the GPS position won't get update from T114 v2. It will only show the position when it get the signal then no update at all.

fifieldt, 2024-12-20:
No need to apologize :) Also, thank you for testing the latest firmware, we appreciate your support :)  

I have a T114, so I will try to confirm the problem.

To be clear, is this what happens: 
1. T114 gets a position the first time
2.  Heltec V3, and the app, can see the position of the T114
3. T114 never updates position again
?

b8b8, 2024-12-20:
How far of a distance did you move between tests?
What app are you using to see the new location packets?