meshtastic/firmware #5616: [Feature Request]: DS18B20 Sensors for Telemetry

SchmalzTech, 2024-12-19:
### Platform

Cross-Platform

### Description

I have a ton of DS18B20 one-wire temperature sensors lying around, device type is 0x28. At one time these were very commonly used, so I am surprised that it wasn't the first thing on the supported list in the telemetry section. They were made by Dallas Semiconductor, now part of Maxim. The part numbers may have changed, but that was the Dallas part number.  

Not sure how hard this is to implement, but it would be great if I could slap these on some nodes and collect data from them.

garthvh, 2024-12-19:
Sensors need to be I2c

closed 2024-12-19 by garthvh

SchmalzTech, 2024-12-19:
> Sensors need to be I2c

Am I mistaken thinking that onewire == I2c?

fifieldt, 2024-12-20:
I2C has two wires.

There's an interesting discussion regarding this and your sensor at https://forum.arduino.cc/t/get-onewire-reading-through-i2c/982095

SchmalzTech, 2024-12-20:
Yeah, thanks for jogging my memory. It must be whatever hackery was going on in the Arduino forums or wherever at the time I was buying these that put that in my brain, maybe using the SDA without the clock somehow. Might have been a (bad) thing to do at one time.