meshtastic/firmware #5618: [Bug]: Range test module doesn't work on Linux-Native Devices (Raspberry Pi HAT)

SteveHawk, 2024-12-20:
### Category

Other

### Hardware

Linux Native

### Firmware Version

2.5.16

### Description

Range Test module on Linux native devices doesn't seem to be working. I'm using a Raspberry Pi 4 with a WaveShare LoRA HAT, and accessing from the web ui.

When range test module is enabled (with or without sending), the node can neither receive nor send range test packets from/to other nodes in the mesh.

I previously installed 2.5.1 and it didn't work back then. On the latest version it still doesn't work.

SteveHawk, 2024-12-20:
Sure thing.

Btw I encountered another bug, when I set `LogLevel: debug` in config.yaml, meshtasticd (2.5.16) crashed. It started fine in `info` level.

```shell
steve@raspberrypi /e/meshtasticd> meshtasticd
Portduino is starting, VFS root at /home/steve/.portduino/default
Set up Meshtastic on Portduino...
Using local config.yaml as config file
Also using "/etc/meshtasticd/config.d/lora-waveshare-sxxx.yaml" as additional config file
fish: Job 1, 'meshtasticd' terminated by signal SIGSEGV (Address boundary error)
```

So I downgraded to 2.5.15 and got it working. Here's the log after I enabled range test module with 30s message interval in web ui.

```shell
INFO  | 15:21:25 91 [RouPortduino is starting, HWID=1, VFS root at /root/.portduino/default
Set up Meshtastic on Portduino...
Using /etc/meshtasticd/config.yaml as config file
Also using "/etc/meshtasticd/config.d/lora-waveshare-sxxx.yaml" as additional config file
DEBUG | ??:??:?? 0 Upgrade time to quality NTP
DEBUG | 15:21:32 0 Read RTC time as 1734708092
INFO  | 15:21:32 0

//\ E S H T /\ S T / C

DEBUG | 15:21:32 0 Filesystem files:
DEBUG | 15:21:32 0  /.. (0 Bytes)
DEBUG | 15:21:32 0  /prefs/channels.proto (94 Bytes)
DEBUG | 15:21:32 0  /prefs/db.proto (1202 Bytes)
DEBUG | 15:21:32 0  /prefs/.. (0 Bytes)
DEBUG | 15:21:32 0  /prefs/. (0 Bytes)
DEBUG | 15:21:32 0  /prefs/config.proto (195 Bytes)
DEBUG | 15:21:32 0  /prefs/module.proto (96 Bytes)
DEBUG | 15:21:32 0  /. (0 Bytes)
INFO  | 15:21:32 0 No I2C device configured, Skip
INFO  | 15:21:32 0 No I2C devices found
INFO  | 15:21:32 0 S:B:37,2.5.15.79da236
INFO  | 15:21:32 0 Init NodeDB
INFO  | 15:21:32 0 Load /prefs/db.proto
INFO  | 15:21:32 0 Loaded /prefs/db.proto successfully
INFO  | 15:21:32 0 Loaded saved devicestate version 22, with nodecount: 5
INFO  | 15:21:32 0 Load /prefs/config.proto
INFO  | 15:21:32 0 Loaded /prefs/config.proto successfully
INFO  | 15:21:32 0 Loaded saved config version 22
INFO  | 15:21:32 0 Load /prefs/module.proto
INFO  | 15:21:32 0 Loaded /prefs/module.proto successfully
INFO  | 15:21:32 0 Loaded saved moduleConfig version 23
INFO  | 15:21:32 0 Load /prefs/channels.proto
INFO  | 15:21:32 0 Loaded /prefs/channels.proto successfully
INFO  | 15:21:32 0 Loaded saved channelFile version 22
DEBUG | 15:21:32 0 cleanupMeshDB purged 0 entries
DEBUG | 15:21:32 0 Use nodenum 0x323a5147
INFO  | 15:21:32 0 Wanted region 4, using CN
DEBUG | 15:21:32 0 Use compiled/slipstreamed tzplaceholder
DEBUG | 15:21:32 0 Set Timezone to GMT0
DEBUG | 15:21:32 0 Read RTC time as 1734708092
DEBUG | 15:21:32 0 Run without GPS
/root/.portduino/default/prefs/cannedConf.proto does not exist
ERROR | 15:21:32 0 Could not open / read /prefs/cannedConf.proto
INFO  | 15:21:32 0 CannedMessageModule is enabled
DEBUG | 15:21:32 0 Activate sx1262 radio on SPI port /dev/spidev0.0
DEBUG | 15:21:32 0 SX126xInterface(cs=21, irq=16, rst=18, busy=20)
DEBUG | 15:21:32 0 SX126X_DIO3_TCXO_VOLTAGE not defined, not using DIO3 as TCXO reference voltage
INFO  | 15:21:32 0 Start meshradio init
INFO  | 15:21:32 0 Radio freq=489.375, config.lora.frequency_offset=0.000
INFO  | 15:21:32 0 Set radio: region=CN, name=<redacted>, config=0, ch=77, power=19
INFO  | 15:21:32 0 myRegion->freqStart -> myRegion->freqEnd: 470.000000 -> 510.000000 (40.000000 MHz)
INFO  | 15:21:32 0 numChannels: 160 x 250.000kHz
INFO  | 15:21:32 0 channel_num: 78
[... 42 lines cut ...]
DEBUG | 15:22:17 45 [DeviceTelemetry] No modules interested in portnum=67, src=LOCAL
DEBUG | 15:22:17 45 [DeviceTelemetry] Add packet record (id=0x7e516d7e fr=0x47 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 rxtime=1734708137 priority=10)
DEBUG | 15:22:17 45 [DeviceTelemetry] Use AES256 key!
DEBUG | 15:22:17 45 [DeviceTelemetry] enqueue for send (id=0x7e516d7e fr=0x47 to=0xff, WantAck=0, HopLim=3 Ch=0x2c encrypted rxtime=1734708137 hopStart=3 priority=10)
DEBUG | 15:22:17 45 [DeviceTelemetry] txGood=1,txRelay=0,rxGood=0,rxBad=0
DEBUG | 15:22:17 45 [DeviceTelemetry] Use channel 0 (hash 0x2c)
DEBUG | 15:22:17 45 [DeviceTelemetry] Use AES256 key!
DEBUG | 15:22:17 45 [DeviceTelemetry] decoded message (id=0x7e516d7e fr=0x47 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 rxtime=1734708137 hopStart=3 priority=10)
DEBUG | 15:22:17 45 [RadioIf] Started Tx (id=0x7e516d7e fr=0x47 to=0xff, WantAck=0, HopLim=3 Ch=0x2c encrypted rxtime=1734708137 hopStart=3 priority=10)
DEBUG | 15:22:17 45 [RadioIf] Packet TX: 591ms
DEBUG | 15:22:17 45 [RadioIf] Completed sending (id=0x7e516d7e fr=0x47 to=0xff, WantAck=0, HopLim=3 Ch=0x2c encrypted rxtime=1734708137 hopStart=3 priority=10)
```

The thing is that, there's just no log about range test at all... I waited for a few minutes, and not seeing it at all, and my other nodes also does not receive any range test packets.

I can also verify the settings are indeed in effect using `meshtastic` CLI tool:

```shell
(meshtastic-venv) steve@raspberrypi /e/meshtasticd> meshtastic --get range_test
No Serial Meshtastic device detected, attempting TCP connection on localhost.
Connected to radio
range_test.enabled: True
range_test.sender: 30
Completed getting preferences
```

GUVWAF, 2024-12-20:
The RangeTestModule is still included only for ESP32 and nRF52: 
https://github.com/meshtastic/firmware/blob/e1de439a7f7c132e469ac2ed32e9b90ad527c3e3/src/modules/RangeTestModule.cpp#L33 

Not sure if there's still a reason why it won't work on Portduino or RP2xxx, but it seems support has not been added yet.

SteveHawk, 2024-12-20:
Oh, that explains.

Now that it's not a bug, can we convert this issue into a feature request? Range test is a pretty useful feature, and feels like it can be hardware independent? (correct me if im wrong), wish it could be supported on native Linux (portduino) platform.

closed 2025-03-21 by thebentern