meshtastic/firmware #5631: [Bug]: tdeck and app chrashing

canadamadman, 2024-12-21:
### Category

BLE, Other

### Hardware

T-Deck

### Firmware Version

2.5.15.79da236

### Description

the tdeck plus dose not want to stay connected if the ringtone is set to nothing or incorrect tone, ble refuse to work, then device resets. the app also crashes when trying to access the external notifications after ringtone is set and used. this seem to be just with the tdeck. also if connected to usb to use serial monitor while doing this makes it worse. docent reset after settings changes as well.

fifieldt, 2024-12-23:
Did you get any logs when connected to the serial monitor? They'd really help us to work out what's going on.

canadamadman, 2024-12-23:
[meshtastic-log-2024-12-23T23-48-57.361Z.log](https://github.com/user-attachments/files/18234189/meshtastic-log-2024-12-23T23-48-57.361Z.log)

canadamadman, 2024-12-23:
that log is from turning on state to going in to settings trying to access "external notifications" app crashes. will flash fresh and send 2nd log to show the issue of no ble once changed notification plays, it break(?) the ble

canadamadman, 2024-12-24:
[meshtastic-log-2024-12-24T00-02-05.244Z.log](https://github.com/user-attachments/files/18234252/meshtastic-log-2024-12-24T00-02-05.244Z.log)

this second log is when i change the "ring tone", the device dose not reboot. and if you recieve a message that would trigger the tone, causes ?(ble failure)? and app says device is sleeping. also cant access settings.

garthvh, 2024-12-24:
Power saving is on by default which will put BLE to sleep and you have to wake it back up by pressing a key or button, try turning off power saving

canadamadman, 2024-12-24:
It dosent wake up though. It stays in that state. The device itself seems to work "normally" but cant connect

canadamadman, 2024-12-24:
i am able to recreate this issue every time, from full erase flash. i have also discover that after the app says the device is asleep it completely disconnects. the t-deck itself can send messages with the keyboard but app wont connect. sorry just trying to give as much information as possible.

nasiralamreeki, 2025-02-27:
Are you still having this issue on the latest release?

fifieldt, 2025-02-27:
As @nasiralamreeki says, there is a new firmware preview for T-Deck that is a significant improvement, I recommend trying it out: https://meshtastic.org/blog/meshtastic-2-6-preview/ !

canadamadman, 2025-02-28:
got to be honest thought this was closed and "abandon" as there was never a actual response that helped. i just stopped using my t decks and moved on to a different device.. and i cant use 2.6 as I'm not smart enough to get to it.