meshtastic/firmware #5632: [Bug]: bootloop on esp-32 with MQTT enabled

fzellini, 2024-12-21:
### Category

WiFi

### Hardware

Heltec V3

### Firmware Version

2.5.17

### Description

device crash and restart loop when wifi and mqtt are enabled.

### Relevant log output

```Shell
INFO  | ??:??:?? 1 Set RX gain to boosted mode; result: 0
INFO  | ??:??:?? 1 SX1262 init success
DEBUG | ??:??:?? 1 Init MQTT

abort() was called at PC 0x401eba13 on core 1

Backtrace: 0x40084139:0x3ffd7ea0 0x4009768d:0x3ffd7ec0 0x4009d4c1:0x3ffd7ee0 0x401eba13:0x3ffd7f60 0x401eba5a:0x3ffd7f80 0x401ebd8f:0x3ffd7fa0 0x401fc6af:0x3ffd7fc0 0x401080de:0x3ffd7fe0 0x4010c191:0x3ffd8020 0x40137a09:0x3ffd8100 0x400e8103:0x3ffd82f0
```

GUVWAF, 2024-12-21:
What is the exact commit you used? 2.5.17 is not released yet.

fzellini, 2024-12-21:
I think the very last, pulled two hours ago

fifieldt, 2024-12-21:
This was fixed.

closed 2024-12-21 by fifieldt