meshtastic/firmware #5637: [Bug]: node doesn't wake up after running out of battery

ebaschiera, 2024-12-21:
### Category

Hardware Compatibility

### Hardware

Heltec Mesh Node T114

### Firmware Version

2.5.14.f2ee0df

### Description

Hi, I connected a solar panel and 3 18650 cells in parallel to a Heltec T114 node. 
When batteries run out of energy, the node switches off (I think), and when the sun rises, the node doesn't wake up.
I was expecting the node to turn itself on. This happened several times, even after a couple of hours under a bright sun. The orange led is on, telling there's incoming power from the solar panel, but the node remains off.
The only way to wake it up is to disconnect both batteries and solar panel and reconnect them back.
I tried to thinker on the power saving configuration, but it doesn't seem to do what it says.
Thanks!

fifieldt, 2024-12-21:
Hi, thanks for writing in.

This is unfortunately a known issue with NRF52-based hardware. You can find other discussions about this issue with the rak4631:

https://meshtastic.discourse.group/t/rak-solar-node-automatic-restart-after-brownout-upon-eventual-solar-re-charge/13196
https://meshtastic.discourse.group/t/rak-4631-solar-repeater/7694

So, you'd need to find a hardware solution.

Since we can't fix this in firmware, closing this one with our apologies.

closed 2024-12-21 by fifieldt

ebaschiera, 2024-12-22:
Thank you for the links.
No need to apologize :)