meshtastic/firmware #5638: [Board]: DIY ESP32S3 ST7798 XPT2046 (stand alone TFT lora node)

steefst33f, 2024-12-21:
### SOC

ESP32

### Lora IC

Sx1262 Lora Node (HF) WaveShare (868MHz)

### Product Link

https://github.com/steefst33f/DIY_Esp32S3_ST7789_XPT2046_LoRa_Node

### Description

Well it's just a DIY stand alone Lora node not a real new product. 

I just wanted to make a cheap Lora node with a 320x240 touch display and keyboard to be able to play with the experimental TFT UI for Meshtastic. 
I already wrote the code to add it as a DIY node (PRIVATE_HW) in my own fork of Meshtastic firmware and ui-device repositories.  (https://github.com/steefst33f/meshtastic-device-ui & https://github.com/steefst33f/meshtastic-firmware)
But I'm new to contributing to a GitHub project and didn't manage to make a pull request from my fork the original Meshtastic repository.

This DIY board is made out of the following components:
- ESP32S3 N16R8 
- TFT Display (320x240) ST7789 driver
- Resistive touch XPT2046 driver
- Lora Sx1262 (868Mhz)
- GNNS NEO-6M (Blox)
- keyboard M5CardKB

mverch67, 2024-12-21:
I think the community has not much interest in merging proprietary diy code for devices that don't have larger community support and additional benefits (such as new platforms). Also device-ui provides a generic class LGFX_GENERIC.h that allows to add TFT devices without a single change of the device-ui library. 

You can find an existing and working example for a community supported diy TFT device using esp32-s3, tft display with ST7789 / ILI9341 / ILI9388 driver and XPT2046 / FT6236 here: https://github.com/valzzu/Mesh-Tab
See also the integration via LGFX_GENERIC: 
https://github.com/meshtastic/firmware/blob/tft-gui-work/variants/mesh-tab/platformio.ini

steefst33f, 2024-12-22:
Ah shit ok I understand, I see I have been working in parallel of this. I didn't see there was already one when I started working on it beginning of November. Otherwise I wouldn't have gone through the trouble. 😅 I used the T-Deck code as a template that was using its own driver, I will take look at the LGFX_GENERIC, thanks.

JohnTaipei, 2024-12-22:
Hello Sir:
May I have your firmware??

closed 2025-01-03 by mverch67

PJ5168, 2025-03-10:
> I think the community has not much interest in merging proprietary diy code for devices that don't have larger community support and additional benefits (such as new platforms). Also device-ui [...]

Dear Sir,
Tried to build the mesh-tab. All the enivs failed. The build process tried to build the bluetooth which should be excluded.