meshtastic/firmware #5650: [Bug]: Erase fw v3 for rak4631 in does not work on SoftDevice S140 6.1.1

recrof, 2024-12-22:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.5.x

### Description

Meshtastic_nRF52_factory_erase_v3_S140_6.1.0.uf2 that's [bundled in every nrf52840 github release from version 2.5.0](https://github.com/meshtastic/firmware/commit/3bb1cb8f1de3eb56e4dff01c0592651c50155064), does not erase flash fs on rak4631 when using meshtastic-cli with `--noproto` flag. 
Serial connects but hangs without any output - not erasing the flash. [Erase firmware v2 that downloaded from flasher.meshtastic.org](https://flasher.meshtastic.org/uf2/nrf_erase2.uf2) (`nrf_erase2.uf2`) works with [guide in docs](https://meshtastic.org/docs/getting-started/flashing-firmware/nrf52/nrf52-erase/)

fifieldt, 2024-12-22:
Thanks for that. I believe we upgraded the NRF devices to SoftDevice 7 at some point, so Meshtastic_nRF52_factory_erase_v3_S140_7.3.0.uf2 would be the one to use now. It looks like that's also part of the bundle?

caveman99, 2024-12-23:
the 4631 stays on 6.1.1 - but the file should just work if you follow the instrustions. i just confirmed with a 4631 on my desk.

recrof, 2024-12-23:
> the 4631 stays on 6.1.1 - but the file should just work if you follow the instrustions. i just confirmed with a 4631 on my desk.

I believe that the instructions are pretty easy to follow. might be flash corruption(the reason I tried to wipe the device) reason to not work?

fifieldt, 2024-12-23:
@markbirss - ideas?

markbirss, 2024-12-24:
> > the 4631 stays on 6.1.1 - but the file should just work if you follow the instrustions. i just confirmed with a 4631 on my desk.
> 
> I believe that the instructions are pretty easy to follow. might be flash corruption(the reason I tried to wipe the device) reason to not work?

To confirm bad flash a full nrf52 erase with swdio using pyocd or openocd is required

pyocd is not as good at reporting flash error but openocd does as example below
[image]

Discord Link (using picoprobe and pyocd) notes
https://discord.com/channels/867578229534359593/871540140382761020/1252622025269448714

closed 2025-03-03 by github-actions[bot]