meshtastic/firmware #5654: [Feature Request]:

MALAONE1, 2024-12-23:
### Platform

NRF52

### Description

Would it be possible to add an i2c rain sensor? This should not conflict with other sensors with the I2C address. Is something like that planned? It would complete the sensor diversity.

caveman99, 2024-12-23:
Sounds good. Do you have a particular one in mind? 'Rain sensor' could be two things. Like a rain gauge, or a raindrop detector...

MALAONE1, 2024-12-23:
@caveman99 this one for example, I'm already using it with a Lora APRS WX station and it works very well. The module that is adapted to I2C is included.https://www.dfrobot.com/product-2689.html?srsltid=AfmBOoqFk20aY0QuZ24DKvQSubfwLxWmFfCtuRq9tfLDscwlsPhMssh_

caveman99, 2024-12-23:
Nice, i ordered that two hours ago ;-)

MALAONE1, 2024-12-23:
@caveman99 Perfect 😉👍

fifieldt, 2025-03-02:
@caveman99 - still interested in this one?

caveman99, 2025-03-02:
Sensor is implemented since 2.5.17 ;-)

closed 2025-03-02 by fifieldt