meshtastic/firmware #5656: [Bug]: T1000e losing programing

madeofstown, 2024-12-23:
### Category

Other

### Hardware

Seeed Card Tracker T1000-E

### Firmware Version

2.5.16 and 2.5.15

### Description

I really like the T1000e for it's features and battery life but there is something strange going on. Within the last 2 weeks (with current firmware 2.5.16 as well as 2.5.15) I've had the T1000e lose its programming over night while charging 2 times. I've also noticed that it will restart itself a few times during the day playing the telltale startup tune on its buzzer.  Because of the time when this is happening (overnight while I'm sleeping) I am not able to provide any logs.

thebentern, 2024-12-23:
Please update us if you are able to get some logs.
A couple of questions with setup, do you use MQTT w/ client proxy or are you on a busy mesh (i.e. >100 nodes)?

madeofstown, 2024-12-23:
Quick update. First, not using MQTT at all but my node database usually sits at 80 nodes. 

The first time I lost programming about a week ago it seemed to be just the node database (name was reset but most settings stayed) that was wiped. When it happened again last night everything is now gone.

garthvh, 2024-12-23:
Are you connected via BLE to an android phone?

madeofstown, 2024-12-23:
> Are you connected via BLE to an android phone?

Yes, this is my on-person node so I try to make sure it's always connected to my phone.

todd-herbert, 2024-12-23:
I believe I'd heard @erayd mention that his T1000E had once run out of filesystem space, even after the Node DB size was dropped to 80.

madeofstown, 2024-12-23:
> I believe I'd heard @erayd mention that his T1000E had once run out of filesystem space, even after the Node DB size was dropped to 80.

I've also experienced losing programing several times on a RAK4631 solar node that I've been testing. I wonder if 80 nodes in the DB is still too many for nRF based nodes

garthvh, 2024-12-23:
> I've also experienced losing programing several times on a RAK4631 solar node that I've been testing. I wonder if 80 nodes in the DB is still too many for nRF based nodes

There is an Android BLE disconnect when out of range bug that seems to be the source of the lost user proto that we have not been able to track down.  I have seen the memory while with a device using iOS but only when on the root topic of mqtt for 24 hours +

madeofstown, 2024-12-23:
> > > I believe I'd heard @erayd mention that his T1000E had once run out of filesystem space, even after the Node DB size was dropped to 80.
> > 
> > 
> > I've also experienced losing programing several times on a RAK4631 solar node that I've been testing. I wonder if 80 nodes in the DB is still too many for nRF based nodes
> 
> There is an Android BLE disconnect when out of range bug that seems to be the source of the lost user proto that we have not been able to track down. I have seen the memory wipe while with a device using iOS but only when on the root topic of mqtt for 24 hours +

Interesting, I don't think that's my problem since I keep the T1000e clipped to my shirt and my phone in my front pocket. I do notice that I occasionally need to reconnect the node during the day but I didn't think it was a range issue.

b8b8, 2024-12-23:
Maybe not related specifically but i just had the T1000E GPS lockup for about 13 hours over night and refuse to get GPS lock after running fine for days.

b8b8, 2024-12-27:
Im having a RAK have a hard time maintaining its name as well as it just randomly wiped its memory today. It is a "router" node but has been set to client_mute for the last few weeks since its off the mountain right now. I upgraded to 2.5.17 this morning and started having the issues. This seems to be similar to the T1000E problem. This node was NOT CONNECTED to any phones for the last month.

Firecul, 2024-12-31:
I haven't experienced that myself and I am going out of range of my T1000e frequently. I do however keep my node list quite empty. If I haven't seen a node in a few days I'll normally clear it off. It's possible that has protected me from this bug.

b8b8, 2024-12-31:
> I haven't experienced that myself and I am going out of range of my T1000e (2.5.15) frequently. I do however keep my node list quite empty. If I haven't seen a node in a few days I'll normally clear it off. It's possible that has protected me from this bug.

What phone are you using? It seems to be more prevalent on Android, specifically Pixel phones.

Firecul, 2025-01-01:
> What phone are you using? It seems to be more prevalent on Android, specifically Pixel phones.

I've been switching between my Pixel 6 Pro and iPhone SE 2020 to use with that node for a few months. I've been running it more with the iPhone as I can more easily grab the metric logs from there while I'm experimenting.

ejosterberg, 2025-01-12:
Need to add a me too here. Version 2.5.15.79da236, updated 6 days ago, Lost config twice. Android client, My primary carry around node. 100 nodes in list. Seems to be fine, but then I'll be prompted for the bluetooth paring pin on my cell phone. I'll enter the pin and discover the T1000e is empty. I keep the backup handy and I'm up and running in a few minutes, but twice in 6 days. Was working for months on a v2.4 version without problems.

ejosterberg, 2025-01-12:
And yes, using a Pixel 9ProXL Android version 15, Build AP4A.241205.013 System update December 5th 2024

phaseloop, 2025-01-12:
I'll chime in - Heltec V3 - v2.5.15.79da236.

I shut it down by long pressing the key and the next morning I woke it up (one short key press) and both node db and user name config were gone.

I use it with Android BLE. Before shutting down phone lost connection with Heltec during some operations - I was sending some messages and BT lost range. Now app still shows some messages as waiting for ACK even after 12h) - and for some reason now does not display previous battery graph (history log below is fine).

esev, 2025-01-23:
I suspect this is related to a flash write failing when there is a BLE disconnect at the same time. We may have inadvertently introduced a second issue in 2.5.19. More details in #5839.

thebentern, 2025-01-23:
Closing this one as well to continue consolidation of tracking in #5839

closed 2025-01-23 by thebentern