meshtastic/firmware #5669: [Feature Request]: Esp32 CPU temprature

Noworries23, 2024-12-26:
### Platform

ESP32

### Description

ESP32 boards support CPU temperature monitoring. 
https://www.espboards.dev/blog/esp32-inbuilt-temperature-sensor/#google_vignette 

Could be useful to incorporate CPU monitoring along with the battery/airtime monitoring that already exists for reliability/safety

caveman99, 2024-12-26:
duplicate of https://github.com/meshtastic/firmware/issues/3883

closed 2024-12-26 by caveman99

Noworries23, 2024-12-26:
Sorry if i misinterpreted but isn't [#3883](https://github.com/meshtastic/firmware/issues/3883#issue-2294537269) to measure external/ambient temperature? 
I was looking to monitor the CPU temperature of the board itself, unrelated to the ambient temperature. Thanks.

rob-fi, 2025-08-07:
> Sorry if i misinterpreted but isn't [#3883](https://github.com/meshtastic/firmware/issues/3883#issue-2294537269) to measure external/ambient temperature? I was looking to monitor the CPU temperature of the board itself, unrelated to the ambient temperature. Thanks.

I also believe it was misinterpreted as a duplicate of the other issue. Came here as I'm looking to debug possible ESP32 overheating in sunlight.