meshtastic/firmware #5684: Remove legacy remote node administration

fifieldt, 2024-12-27:
Firmware 2.5 introduced Public Key Cryptography (PKC) to Meshtastic. With it came the ability for node operators to specify which keys should be allowed to remotely administer a node. This was a major security improvement - previously node operators had to have a single management channel and one pre-shared key to secure their nodes.

Until now, we've kept the 'legacy' functionality in place, so users of version 2.5+ firmware could still manage older nodes. Now firmware 2.5+ is in widespread use, and the new remote node administration functionality is well-tested, we can safely remove 'legacy' remote node administration. This has benefits of a smaller and more maintainable code-base, reducing a potential vulnerability vector.

References:
https://meshtastic.org/docs/overview/encryption/
https://meshtastic.org/docs/development/reference/encryption-technical/
https://meshtastic.org/docs/configuration/remote-admin/

garthvh, 2025-01-09:
> This new way of remote admin is not working on nodes where licensed ham radio is enabled. Only the older way with admin channel still works on nodes for licensed radio operators. I tested multiple times, last time on 2.5.15. So removing the legacy functionality will make remote nodes used by licensed ham operators unavailable for remote 

an admin channel that is not encrypted is not a supported option, the legacy admin channel is not expected to work in licensed mode.

iondulgheru, 2025-01-09:
> an admin channel that is not encrypted is not a supported option, the legacy admin channel is not expected to work in licensed mode.

Yet, it works. Tested on both remote node and active node on 2.5.15.
And the new remote admin method does not work.

iondulgheru, 2025-01-10:
So how can I change the configuration on remote nodes after the the admin channels are removed?
The new remote administration method does not work if "Licensed amateur radio" is enabled.

ianmcorvidae, 2025-01-10:
There isn't a way to safely and legally change configurations on remote nodes along with the licensed amateur radio setting. Either you do it by getting close enough to connect with BLE/TCP/Serial, you encrypt it (breaking license rules), or you add an unencrypted admin channel that anyone could trivially use to change your nodes' configurations (though I haven't checked, we may not allow an unencrypted admin channel either).

The best workaround I can think of without spending too much time thinking about it would be to create some sort of a side-channel that uses the unlicensed side of things and talks to the licensed-mode node over serial to change configuration. But that would probably mean 2 additional MCUs as well.

mntn-xyz, 2025-01-31:
Signed plaintext messages do not appear to break license rules, as the signature does not obfuscate the information sent. So in theory something could be built to do basic remote admin for these users using signatures. I think this may be unrealistic though due to the extra length a signature would add to the packet.