meshtastic/firmware #5700: [Feature Request]: Only allow empty keys for nodes that are set as a licensed user

garthvh, 2024-12-30:
### Platform

Cross-Platform

### Description

Right now we allow users who are not licensed users to set empty keys for channels when they are not in licensed user mode, we should restrict empty keys to only licensed users so that Hams do not have to be concerned with potentially rebroadcasting messages sent from unlicensed users.

mepholic, 2024-12-31:
I think this is suggesting that non-HAM licensed nodes should _not_ be allowed to use blank keys, or be able to send completely unencrypted traffic. It seems like this is being proposed as a "solution" for HAMs who are worried about digipeating traffic originating from nodes run by unlicensed operators.

I do not think that usage of encryption should be used as a heuristic to decide whether or not the traffic originated from a HAM operator or not, and in addition to this, nothing stops someone from lying about their licensing status. If the purpose of this feature request is to allow a HAM operator to selectively rebroadcast traffic from other HAM operators only, another field should probably be added to the packets to indicate station license status, and then an option should be added to the Rebroadcast Mode dropdown for "only packets from licensed stations", or something like that.

mepholic, 2024-12-31:
I had some additional musings in Discord, which I'll include here:

> is anyone here familiar with the technical aspects of the LoTW PKI? would it be possible to sign meshtastic frames with your callsign cert and have others be able to verify the signature?

> _if_ that is possible, one disadvantage of it would be that it would put ham verification authority for meshtastic users pretty much solely in the hands of the ARRL, but on the flip side, their direct mail authorization and the technical aspects of the PKI would allow folks to be fairly certain that signed packets are from actual hams, and in addition to that, many hams already have LoTW certs

mepholic, 2024-12-31:
I've given this some more thought, and I realized that requiring encryption on unlicensed traffic means that Meshtastic would be outright illegal to use to jurisdictions that ban encryption, unless users in those jurisdictions turn on licensed operator mode, which in turn just enables even more foot guns. Is this really a good approach?

dm5tt, 2024-12-31:
Isn't there an option to still use some kind of cryptographic signing without encryption enabled? Encryption ham radio is forbidden globally - signing not.

thebentern, 2025-01-01:
> I've given this some more thought, and I realized that requiring encryption on unlicensed traffic means that Meshtastic would be outright illegal to use to jurisdictions that ban encryption, unless users in those jurisdictions turn on licensed operator mode, which in turn just enables even more foot guns. Is this _really_ a good solution? 🤔

Areas that do not allow any encryption may be an even smaller audience of meshtastic usage than licensed mode users. I have not seen any examples of folks contending with it. I suspect most individuals subject to those restrictions just pick a lora region and hang out on the defaults or don't use meshtastic at all.

SpudGunMan, 2025-01-04:
It's also very easy to exploit this keyless nature 

Crazy idea 🥔what about some base key where its key=Licensed:Callsign with a very reversible hash which is the encryption but it's not 

There needs to be a way to route these packets effectively over all networks to make this enjoyable and help in its emergency use in a way where it's not really simple and useful for others to just drop encryption (why wouldn't everyone use nokey if the ham infrastructure is built robust in an area, or to bypass duty cycle limits)

garthvh, 2025-02-14:
Will be released with 2.6

closed 2025-02-14 by garthvh

thebentern, 2025-02-14:
Fixed by #6047