meshtastic/firmware #5701: 🐞 [Bug]: Primary Device Should Never Record/Display Self Reported SNR/RSSI

Xaositek, 2024-12-30:
### Firmware Version

2.5.18

### What did you do?

@thebentern and I were discussing in Discord and while I am unsure the origin of this, it was determined this shouldn't happen.

### Expected Behavior

We should never record or display a self reported SNR/RSSI

### Current Behavior

[image]

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

fifieldt, 2024-12-31:
Q) have you connected this app to other devices that might be able to see this node?

Just wondering whether it's caching that info from a past connection

Xaositek, 2024-12-31:
I was connected to a RAK4631 before swapping to my T1000. These two nodes I do swap between on my iPhone but I don't think they've been on at the same time recently.

garthvh, 2025-01-01:
The only way this can happen is if there is a SNR value for the node > 0 and it is not on MQTT

PR #6793 (2025-05-11): Add some no-nonsense coercion for self-reporting node values
Related to #4618
Closes #5701
closed 2025-05-11 by thebentern