meshtastic/firmware #5702: [Bug]: Static GPS Position will be changed by firmware

RasPix, 2024-12-30:
### Category

Other

### Hardware

T-Lora v2 1.6

### Firmware Version

2.5.15.79da236

### Description

Hello,
I use a TTGo V2 1.6 which do not have a GPS module. I activated the manual Input vor a fixed Position. The values for lat / Long we're entered. After saving and restart we're this values slightly changed by fw and the dislayed Position ist far away from real Position. I tried this Sevetal Times with slightly modified values but the fw changed this values to different positiones. 
Finally i disabled this function because of the wrong Position in the map.
I guess it looks Like a bug, because fw do not have to Change reasonable User Inputs. (A fittet resolution may be possible but never in this range)
Regads Tom

fifieldt, 2024-12-30:
Hi Tom,

Thanks for using Meshtastic :)

By default, we try and protect your privacy by reducing the precision of position information. If you'd like to have an exact position instead, just modify the position precision settings according to your needs:
https://meshtastic.org/docs/configuration/radio/channels/#position-precision

-Tom

closed 2024-12-31 by fifieldt

kzkvv, 2025-11-27:
Hello, I met the same issue today. Found this issue and looks like it will be a common problem. Maybe it will be helpful to mention 'position precision' functionality somewhere on 'Position' page in settings.