meshtastic/firmware #5747: [Bug]: Setting the fixed GPS position is not possible via Python CLI

rena2019, 2025-01-05:
### Category

Other

### Hardware

Heltec Wireless Paper

### Firmware Version

2.5.15.79da236

### Description

I'm trying to configure my meshtastic device via Python CLI only (without Android device). But after setting the GPS position with

```
meshtastic --port COM5 --setlat 53.3XXXX --setlon 9.8XXXX
  Connected to radio
  Fixing latitude at 53.3XXXX degrees
  Fixing longitude at 9.8XXXX degrees
  Setting device position and enabling fixed position setting
```

it's not possible to find the device at https://meshmap.net/ 
and the following warning is displayed in the serial log:

DEBUG | 14:56:24 403 [Power] Battery: usbPower=0, isCharging=0, batMv=3867, batPct=67
DEBUG | 14:56:29 408 [Screen] determineMode(): 
DEBUG | 14:56:29 408 [Screen] refresh=SKIPPED, reason=FRAME_MATCHED_PREVIOUS, frameFlags=0x1
WARN  | 14:56:41 420 [mqtt] MQTT Map report enabled, but no position available

Talie5in, 2025-01-07:
As per the tip on https://meshtastic.org/docs/configuration/radio/position/

Because the device will reboot after each command is sent via CLI, it is recommended when setting multiple values in a config section that commands be chained together as one. This is especially important for position values to ensure they are set at the same time and avoid being overwritten by subsequent commands.

```Example:
meshtastic --set position.fixed_position true --setlat 37.8651 --setlon -119.5383```

ianmcorvidae, 2025-01-07:
The current `--setlat`, `--setlon` `--setalt` commands use an admin message that also enables fixed position (though all of the relevant values among the three should be set at once). However, the first question to ask is if you've already set fixed position to true before running the `--setlat` etc. commands; if you have, they won't take.

One way to test this is to use `meshtastic --remove-position` first to fully clear any fixed position and turn off the setting, allow the node to reboot, and then try setting it again with `meshtastic --setlat XX.XXX --setlon YY.YYY`

closed 2025-03-17 by github-actions[bot]