meshtastic/firmware #5754: [Bug]: RAK4631 stops updating after a while (and only comes back after any kind of serial connection)

andrepcg, 2025-01-05:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.5.15.79da236

### Description

I'm also experiencing some issues with a RAK4631 (i have 2 and one is acting weird).

List of nodes:
- A (esp32), connected to wifi and MQTT
- B (RAK4631), node connected to a Raspberry Pi that I can SSH into. Firmware: `2.5.15"`
- C (RAK4631), purely remote node (solar). Firmware: `2.5.18.89ebafc"`

I'm in a very low density area and only my nodes are present.

`A` can see `C` and somewhat see `B`. `B` can see `C`.

**Looking at the meshtastic map, node `B` stops being updated after a while (some random time).** If I ssh into the raspberry and leave a serial connection opened with `--noproto`, it never stops responding and the "last updated" info in the map is always fresh.

It also seems that triggering any serial connection with the board makes it soft reset and it starts responding again and the map info gets instantaneously updated (by talking to the `A` node).

### 1. Requesting telemetry for node `B` on node `A`:
```
meshtastic --request-telemetry --dest '!5570b7aa'
Connected to radio
Sending device_metrics telemetry request to !5570b7aa on channelIndex:0 (this could take a while)
Aborting due to: Timed out waiting for telemetry
```

### 2. Node `B` logs right after starting `--noproto`:
```
~ $ meshtastic --noproto
WARNING file:mesh_interface.py _sendToRadio line:1117 Not sending packet because protocol use is disabled by noProto
Connected to radio
Mv=4275, batPct=100
DEBUG | 13:58:21 165 [Power] Battery: usbPower=1, isCharging=1, batMv=4277, batPct=100
ERROR | 18:39:17 189821 [RadioIf] Ignore received packet due to error=-7
DEBUG | 18:39:17 189821 [RadioIf] Packet RX (noise?) : 673ms
INFO  | 18:39:17 189822 [NodeInfo] Send our nodeinfo to mesh (wantReplies=0)
DEBUG | 18:39:17 189822 [NodeInfo] cancelSending id=0x2afbc70d, removed=0
INFO  | 18:39:17 189822 [NodeInfo] Send owner !5570b7aa/RUB1-02########/RB02
DEBUG | 18:39:17 189822 [NodeInfo] Partially randomized packet id 3399141136
DEBUG | 18:39:17 189822 [NodeInfo] Update DB node 0x5570b7aa, rx_time=1736015957
DEBUG | 18:39:17 189822 [NodeInfo] handleReceived(LOCAL) (id=0xca9ac710 fr=0x5570b7aa to=0xffffffff, WantAck=0, HopLim=4 Ch=0x0 Portnum=4 rxtime=1736015957 priority=10)
DEBUG | 18:39:17 189822 [NodeInfo] No modules interested in portnum=4, src=LOCAL
DEBUG | 18:39:17 189822 [NodeInfo] Add packet record (id=0xca9ac710 fr=0x5570b7aa to=0xffffffff, WantAck=0, HopLim=4 Ch=0x0 Portnum=4 rxtime=1736015957 priority=10)
DEBUG | 18:39:17 189822 [NodeInfo] Expand short PSK #1
DEBUG | 18:39:17 189822 [NodeInfo] Use AES128 key!
DEBUG | 18:39:17 189822 [NodeInfo] enqueue for send (id=0xca9ac710 fr=0x5570b7aa to=0xffffffff, WantAck=0, HopLim=4 Ch=0x8 encrypted len=100 rxtime=1736015957 hopStart=4 priority=10)
DEBUG | 18:39:17 189822 [NodeInfo] txGood=4,txRelay=1,rxGood=10,rxBad=8
INFO  | 18:39:17 189822 [Position] Send pos@0:6 to mesh (wantReplies=0)
DEBUG | 18:39:17 189822 [Position] cancelSending id=0x4fe0cf0f, removed=0
DEBUG | 18:39:17 189822 [Position] Send location with precision 16
INFO  | 18:39:17 189822 [Position] Strip low RTCQuality (2) time from position
INFO  | 18:39:17 189822 [Position] Position reply: time=0 lat=40196xxxx lon=-8418xxxx
DEBUG | 18:39:17 189822 [Position] Partially randomized packet id 3745994513
.....
```

### 3. Requesting telemetry for node `B` on node `A` after the step above:
```
meshtastic --request-telemetry --dest '!5570b7aa'
Connected to radio
Sending device_metrics telemetry request to !5570b7aa on channelIndex:0 (this could take a while)
Telemetry received:
Battery level: 101.00%
Voltage: 4.28 V
Total channel utilization: 3.15%
Transmit air utilization: 0.15%
Uptime: 189882 s
```

**If I leave `--noproto` running on node `B` (by using the RPi), it never stops responding!** (or if I run any command that talks to the node via serial)

GUVWAF, 2025-01-05:
> Ignore received packet due to error=-7

and 

> rxGood=10,rxBad=8

is suspicious. You are receiving quite some corrupted packets. Can you check if that's from only the ESP32 or the other RAK or from both?

Is B only powered by the RPi USB port? What is its role set to and do you have power saving enabled?

andrepcg, 2025-01-05:
> is suspicious. You are receiving quite some corrupted packets. Can you check if that's from only the ESP32 or the other RAK or from both?

what's the best to test that?

> Is B only powered by the RPi USB port? What is its role set to and do you have power saving enabled?

yes, powered via usb port. Haven't set up any role specifically, just connected it. I have not enabled any power saving setting but will confirm if there's any defaults

GUVWAF, 2025-01-05:
> what's the best to test that?

Turn one of them off, and try to send packets and see if the error appears. Usually this is caused by an issue with the transmitter, but if it's happening for both it might also be an issue at the receiver. 

> yes, powered via usb port. 

Hmm, I'm wondering if it might be a power supply issue. Maybe it's worth trying to connect it just to mains power and see if the issue persists.

andrepcg, 2025-01-05:
Will try to change the power supply (tomorrow).

In the meantime, here's more logs right after running `--noproto` when the down was down for some time:

```
$ meshtastic --noproto
WARNING file:mesh_interface.py _sendToRadio line:1117 Not sending packet because protocol use is disabled by noProto
Connected to radio
?:?? 6 [SerialConsole] Enqueued local (id=0x0d8d15d0 fr=0x00 to=0xaa, WantAck=0, HopLim=4 Ch=0x0 Portnum=6 WANTRESP PKI priority=70)
INFO  | ??:??:?? 23545 [SerialConsole] Disconnect from phone
DEBUG | ??:??:?? 23545 [SerialConsole] PhoneAPI::close()
DEBUG | ??:??:?? 23545 [Router] Add packet record (id=0x0d8d15d0 fr=0x00 to=0xaa, WantAck=0, HopLim=4 Ch=0x0 Portnum=6 WANTRESP PKI priority=70)
DEBUG | ??:??:?? 23545 [Router] handleReceived(REMOTE) (id=0x0d8d15d0 fr=0x00 to=0xaa, WantAck=0, HopLim=4 Ch=0x0 Portnum=6 WANTRESP PKI priority=70)
DEBUG | ??:??:?? 23545 [Router] Module 'Admin' wantsPacket=1
INFO  | ??:??:?? 23545 [Router] Received Admin from=0x0, id=0xd8d15d0, portnum=6, payloadlen=12
INFO  | ??:??:?? 23545 [Router] Handle admin payload 35
INFO  | ??:??:?? 23545 [Router] Client set module config
INFO  | ??:??:?? 23545 [Router] Set module config: Neighbor Info
INFO  | ??:??:?? 23545 [Router] Save changes to disk
DEBUG | ??:??:?? 23545 [Router] Expand short PSK #1
INFO  | ??:??:?? 23545 [Router] Wanted region 3, using EU_868
INFO  | ??:??:?? 23545 [Router] Radio freq=869.525, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 23545 [Router] Set radio: region=EU_868, name=LongFast, config=0, ch=0, power=27
INFO  | ??:??:?? 23545 [Router] myRegion->freqStart -> myRegion->freqEnd: 869.400024 -> 869.650024 (0.250000 MHz)
INFO  | ??:??:?? 23545 [Router] numChannels: 1 x 250.000kHz
INFO  | ??:??:?? 23545 [Router] channel_num: 1
INFO  | ??:??:?? 23545 [Router] frequency: 869.525024
INFO  | ??:??:?? 23545 [Router] Slot time: 77 msec
ERROR | ??:??:?? 23545 [Router] Ignore received packet due to error=-7
DEBUG | ??:??:?? 23545 [Router] Packet RX (noise?) : 673ms
INFO  | ??:??:?? 23545 [Router] Save /prefs/module.proto
INFO  | ??:??:?? 23546 [Router] Reboot in 7 seconds
DEBUG | ??:??:?? 23546 [Router] Initial packet id 2002897172
DEBUG | ??:??:?? 23546 [Router] Partially randomized packet id 284523797
INFO  | ??:??:?? 23546 [Router] Asked module 'Admin' to send a response
DEBUG | ??:??:?? 23546 [Router] Module 'routing' wantsPacket=1
INFO  | ??:??:?? 23546 [Router] Received routing from=0x0, id=0xd8d15d0, portnum=6, payloadlen=12
DEBUG | ??:??:?? 23546 [Router] Routing sniffing (id=0x0d8d15d0 fr=0x00 to=0xaa, WantAck=0, HopLim=4 Ch=0x0 Portnum=6 WANTRESP PKI priority=70)
DEBUG | ??:??:?? 23546 [Router] Module 'routing' considered
DEBUG | ??:??:?? 23546 [Router] Send response (id=0x10f57d15 fr=0xaa to=0xaa, WantAck=0, HopLim=4 Ch=0x0 Portnum=5 requestId=d8d15d0 priority=120)
DEBUG | ??:??:?? 23546 [Router] Update DB node 0x5570b7aa, rx_time=0
DEBUG | ??:??:?? 23546 [Router] Enqueued local (id=0x10f57d15 fr=0xaa to=0xaa, WantAck=0, HopLim=4 Ch=0x0 Portnum=5 requestId=d8d15d0 priority=120)
DEBUG | ??:??:?? 23546 [Router] Rx someone rebroadcasting for us (id=0x10f57d15 fr=0xaa to=0xaa, WantAck=0, HopLim=4 Ch=0x0 Portnum=5 requestId=d8d15d0 priority=120)
DEBUG | ??:??:?? 23546 [Router] Didn't find pending packet
DEBUG | ??:??:?? 23546 [Router] Add packet record (id=0x10f57d15 fr=0xaa to=0xaa, WantAck=0, HopLim=4 Ch=0x0 Portnum=5 requestId=d8d15d0 priority=120)
DEBUG | ??:??:?? 23546 [Router] handleReceived(REMOTE) (id=0x10f57d15 fr=0xaa to=0xaa, WantAck=0, HopLim=4 Ch=0x0 Portnum=5 requestId=d8d15d0 priority=120)
DEBUG | ??:??:?? 23546 [Router] Module 'routing' wantsPacket=1
INFO  | ??:??:?? 23546 [Router] Received routing from=0x5570b7aa, id=0x10f57d15, portnum=5, payloadlen=2
DEBUG | ??:??:?? 23546 [Router] Routing sniffing (id=0x10f57d15 fr=0xaa to=0xaa, WantAck=0, HopLim=4 Ch=0x0 Portnum=5 requestId=d8d15d0 priority=120)
DEBUG | ??:??:?? 23546 [Router] Received a ACK for 0xd8d15d0, stopping retransmissions
DEBUG | ??:??:?? 23546 [Router] Delivering rx packet (id=0x10f57d15 fr=0xaa to=0xaa, WantAck=0, HopLim=4 Ch=0x0 Portnum=5 requestId=d8d15d0 priority=120)
DEBUG | ??:??:?? 23546 [Router] Update DB node 0x5570b7aa, rx_time=0
DEBUG | ??:??:?? 23546 [Router] Forwarding to phone (id=0x10f57d15 fr=0xaa to=0xaa, WantAck=0, HopLim=4 Ch=0x0 Portnum=5 requestId=d8d15d0 priority=120)
DEBUG | ??:??:?? 23546 [Router] Module 'routing' considered
DEBUG | ??:??:?? 23546 [Power] Battery: usbPower=1, isCharging=1, batMv=4261, batPct=100
INFO  | ??:??:?? 23546 [NodeInfo] Send our nodeinfo to mesh (wantReplies=1)
INFO  | ??:??:?? 23546 [NodeInfo] Send owner !5570b7aa/RUB1-02########/RB02
DEBUG | ??:??:?? 23546 [NodeInfo] Partially randomized packet id 830094614
DEBUG | ??:??:?? 23546 [NodeInfo] Update DB node 0x5570b7aa, rx_time=0
DEBUG | ??:??:?? 23546 [NodeInfo] handleReceived(LOCAL) (id=0x317a3d16 fr=0xaa to=0xff, WantAck=0, HopLim=4 Ch=0x0 Portnum=4 WANTRESP priority=10)
DEBUG | ??:??:?? 23546 [NodeInfo] No modules interested in portnum=4, src=LOCAL
DEBUG | ??:??:?? 23546 [NodeInfo] Add packet record (id=0x317a3d16 fr=0xaa to=0xff, WantAck=0, HopLim=4 Ch=0x0 Portnum=4 WANTRESP priority=10)
DEBUG | ??:??:?? 23546 [NodeInfo] Expand short PSK #1
DEBUG | ??:??:?? 23546 [NodeInfo] Use AES128 key!
DEBUG | ??:??:?? 23546 [NodeInfo] enqueue for send (id=0x317a3d16 fr=0xaa to=0xff, WantAck=0, HopLim=4 Ch=0x8 encrypted hopStart=4 priority=10)
DEBUG | ??:??:?? 23546 [NodeInfo] txGood=0,txRelay=0,rxGood=0,rxBad=1
[... cut ...]
WARNING file:mesh_interface.py _sendToRadio line:1117 Not sending packet because protocol use is disabled by noProto
Connected to radio
DEBUG | ??:??:?? 185 [Power] Battery: usbPower=1, isCharging=1, batMv=4282, batPct=100
DEBUG | ??:??:?? 205 [Power] Battery: usbPower=1, isCharging=1, batMv=4287, batPct=100
DEBUG | ??:??:?? 219 [RadioIf] Lora RX (id=0xc69e9a4a fr=0x50 to=0xff, WantAck=0, HopLim=1 Ch=0x8 encrypted rxSNR=-19.75 rxRSSI=-128 via MQTT hopStart=4)
DEBUG | ??:??:?? 219 [RadioIf] Packet RX: 673ms
```

madeofstown, 2025-01-06:
Just wanted to chime in about the -7 errors - I too see these quite often at my house. I have tried switching radios out and moving positions but they still  remain. I'm cueious if these errors can be caused by other protocols using the same spectrum. I do live in a city.

GUVWAF, 2025-01-06:
Yes, those errors may also appear e.g. due to interference, packet collisions or when you're at the edge of reception (poor SNR), but when 8/18 received packets are corrupted in a 3-node mesh, it likely indicates a failure somewhere.

GUVWAF, 2025-01-11:
@andrepcg Do you have any new results to share?

andrepcg, 2025-01-11:
> @andrepcg Do you have any new results to share?

Literally changed the power source just now. Will monitor the results in the next hours

andrepcg, 2025-01-12:
@GUVWAF I think it’s working reliably now. Since yesterday the node has been responsive, I can get telemetry fine and chat with it. It may have been a power issue. The raspberry it was connected to also had a ADSB usb device and maybe the power was too much for both. Still weird that it would never fail if the serial connection was kept online

GUVWAF, 2025-01-12:
Good to know what the issue was. I'm also not sure why it would not fail if the serial connection was kept alive, but maybe the Pi devotes more current to the port in that case.

kstrauser, 2025-01-19:
I'm still seeing this with 2.5.18.89ebafc. I run [a BBS server](https://github.com/kstrauser/frozenbbs) and it routinely freezes after 12 hours or so. If I then run `meshtastic --noproto` in another terminal, it leaps back to life and responds again (including to previous messages users had sent to it). If I leave the serial logging running, it's stable for days on end.

My radio is plugged into a power USB hub so that it's not drawing power directly from its companion Raspberry Pi.

GUVWAF, 2025-01-19:
At least some power banks are "smart" in a way that they cut the power off when the current draw is too low. Maybe the hub does something similar and keeps it alive when there's activity on the data lines.

kstrauser, 2025-01-21:
That’s surely possible. But for posterity, I’ve seen that on 2 different radios attached to 2 different RPis, on any of each Pi’s 4 USB ports.