meshtastic/firmware #5764: [Bug]: Broken pipe when listening over tcp for events from meshtasticd

bmidgley, 2025-01-05:
### Category

Other

### Hardware

Linux Native

### Firmware Version

2.5.15

### Description

If I use the tcp listener from https://github.com/brad28b/meshtastic-cli-receive-text/blob/main/read_messages_tcp.py it gets stuck every once in a while, crashing in a background thread apparently. I can't rescue it and once it crashes, no events will be received.

```
bmidgley@nightvision:~ $ meshtastic --version
2.5.9
bmidgley@nightvision:~ $ python --version
Python 3.11.2
bmidgley@nightvision:~ $ dpkg -l | grep meshtastic
ii  meshtasticd                          2.5.15.79da236                      arm64        Native Linux Meshtastic binary.
```

```
Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 1394, in run
    self.function(*self.args, **self.kwargs)
  File "/home/bmidgley/env/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1062, in callback
    self.sendHeartbeat()
  File "/home/bmidgley/env/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1051, in sendHeartbeat
    self._sendToRadio(p)
  File "/home/bmidgley/env/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1126, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "/home/bmidgley/env/lib/python3.11/site-packages/meshtastic/stream_interface.py", line 120, in _sendToRadioImpl
    self._writeBytes(header + b)
  File "/home/bmidgley/env/lib/python3.11/site-packages/meshtastic/tcp_interface.py", line 79, in _writeBytes
    self.socket.send(b)
BrokenPipeError: [Errno 32] Broken pipe
```

bmidgley, 2025-01-05:
If I knew the exception had happened, I could restart my connection or exit the script. But it keeps running as if nothing is wrong.

bmidgley, 2025-01-06:
Ok I've lost the exact log message, but I believe it was saying it would blank the screen, then crashed and restarted meshtasticd. I was still configured for a 2.8" waveshare screen. I think the blanking operation is dangerous with the configured and not present screen.

Removing the screen from the config is avoiding the crash.

I'm ok if you want to close or combine or whatever since the ticket has taken a turn.

bmidgley, 2025-01-09:
I'm seeing this crash even when the screen is present

migillett, 2025-01-26:
I'm seeing this error too. Here's my log error.
```meshtastic_listener    | Exception in thread Thread-124:
meshtastic_listener    | Traceback (most recent call last):
meshtastic_listener    |   File "/usr/local/lib/python3.13/threading.py", line 1041, in _bootstrap_inner
meshtastic_listener    |     self.run()
meshtastic_listener    |     ~~~~~~~~^^
meshtastic_listener    |   File "/usr/local/lib/python3.13/threading.py", line 1342, in run
meshtastic_listener    |     self.function(*self.args, **self.kwargs)
meshtastic_listener    |     ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^
meshtastic_listener    |   File "/root/.cache/pypoetry/virtualenvs/meshtatic-listener-c8hN_e31-py3.13/lib/python3.13/site-package                            s/meshtastic/mesh_interface.py", line 1063, in callback
meshtastic_listener    |     self.sendHeartbeat()
meshtastic_listener    |     ~~~~~~~~~~~~~~~~~~^^
meshtastic_listener    |   File "/root/.cache/pypoetry/virtualenvs/meshtatic-listener-c8hN_e31-py3.13/lib/python3.13/site-package                            s/meshtastic/mesh_interface.py", line 1052, in sendHeartbeat
meshtastic_listener    |     self._sendToRadio(p)
meshtastic_listener    |     ~~~~~~~~~~~~~~~~~^^^
meshtastic_listener    |   File "/root/.cache/pypoetry/virtualenvs/meshtatic-listener-c8hN_e31-py3.13/lib/python3.13/site-package                            s/meshtastic/mesh_interface.py", line 1127, in _sendToRadio
meshtastic_listener    |     self._sendToRadioImpl(toRadio)
meshtastic_listener    |     ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
meshtastic_listener    |   File "/root/.cache/pypoetry/virtualenvs/meshtatic-listener-c8hN_e31-py3.13/lib/python3.13/site-package                            s/meshtastic/stream_interface.py", line 120, in _sendToRadioImpl
meshtastic_listener    |     self._writeBytes(header + b)
meshtastic_listener    |     ~~~~~~~~~~~~~~~~^^^^^^^^^^^^
meshtastic_listener    |   File "/root/.cache/pypoetry/virtualenvs/meshtatic-listener-c8hN_e31-py3.13/lib/python3.13/site-package                            s/meshtastic/tcp_interface.py", line 79, in _writeBytes
meshtastic_listener    |     self.socket.send(b)
meshtastic_listener    |     ~~~~~~~~~~~~~~~~^^^
meshtastic_listener    | BrokenPipeError: [Errno 32] Broken pipe
```

migillett, 2025-01-26:
A potential answer for this is my host machine going to sleep while the script is running.

daimyo314, 2025-02-12:
Same error, always happens after presenting the Primary and Complete URLs at the end.

`Aborting due to: [Errno 32] Broken pipe
Traceback (most recent call last):
  File "/meshtastic/venv/lib/python3.11/site-packages/meshtastic/__main__.py", line 994, in onConnected
    interface.close()  # after running command then exit
    ^^^^^^^^^^^^^^^^^
  File "/meshtastic/venv/lib/python3.11/site-packages/meshtastic/tcp_interface.py", line 65, in close
    StreamInterface.close(self)
  File "/meshtastic/venv/lib/python3.11/site-packages/meshtastic/stream_interface.py", line 125, in close
    MeshInterface.close(self)
  File "/meshtastic/venv/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 145, in close
    self._sendDisconnect()
  File "/meshtastic/venv/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1102, in _sendDisconnect
    self._sendToRadio(m)
  File "/meshtastic/venv/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1127, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "/meshtastic/venv/lib/python3.11/site-packages/meshtastic/stream_interface.py", line 120, in _sendToRadioImpl
    self._writeBytes(header + b)
  File "/meshtastic/venv/lib/python3.11/site-packages/meshtastic/tcp_interface.py", line 79, in _writeBytes
    self.socket.send(b)
BrokenPipeError: [Errno 32] Broken pipe

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/meshtastic/venv/bin/meshtastic", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/meshtastic/venv/lib/python3.11/site-packages/meshtastic/__main__.py", line 1988, in main
    common()
  File "/meshtastic/venv/lib/python3.11/site-packages/meshtastic/__main__.py", line 1272, in common
    onConnected(client)
  File "/meshtastic/venv/lib/python3.11/site-packages/meshtastic/__main__.py", line 1002, in onConnected
    interface.close()  # close the connection now, so that our app exits
    ^^^^^^^^^^^^^^^^^
  File "/meshtastic/venv/lib/python3.11/site-packages/meshtastic/tcp_interface.py", line 65, in close
    StreamInterface.close(self)
  File "/meshtastic/venv/lib/python3.11/site-packages/meshtastic/stream_interface.py", line 125, in close
    MeshInterface.close(self)
  File "/meshtastic/venv/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 145, in close
    self._sendDisconnect()
  File "/meshtastic/venv/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1102, in _sendDisconnect
    self._sendToRadio(m)
  File "/meshtastic/venv/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1127, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "/meshtastic/venv/lib/python3.11/site-packages/meshtastic/stream_interface.py", line 120, in _sendToRadioImpl
    self._writeBytes(header + b)
  File "/meshtastic/venv/lib/python3.11/site-packages/meshtastic/tcp_interface.py", line 79, in _writeBytes
    self.socket.send(b)
BrokenPipeError: [Errno 32] Broken pipe`

closed 2025-04-15 by github-actions[bot]

steinjo, 2025-07-23:
I have this issue, when running 
`pub.subscribe(onReceive, "meshtastic.receive")`
and the device (Heltec V3) goes down due to low batterie.  Here is the error message

```
Exception in thread Thread-582:
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 1394, in run
    self.function(*self.args, **self.kwargs)
  File "/home/pi/meshtastic_env/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1128, in callback
    self.sendHeartbeat()
  File "/home/pi/meshtastic_env/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1117, in sendHeartbeat
    self._sendToRadio(p)
  File "/home/pi/meshtastic_env/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1192, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "/home/pi/meshtastic_env/lib/python3.11/site-packages/meshtastic/stream_interface.py", line 120, in _sendToRadioImpl
    self._writeBytes(header + b)
  File "/home/pi/meshtastic_env/lib/python3.11/site-packages/meshtastic/tcp_interface.py", line 91, in _writeBytes
    self.socket.send(b)
BrokenPipeError: [Errno 32] Broken pipe

```

marek22k, 2026-01-03:
What is the current status? I am still getting the same error message:
```
└──╼ $meshtastic --host 127.0.0.1:9443 --export-config
Error connecting to 127.0.0.1:9443:[Errno 32] Broken pipe
```