meshtastic/firmware #5781: [Bug]: Power Resuming after deep sleep

mentanah, 2025-01-07:
### Category

Hardware Compatibility

### Hardware

T-Beam

### Firmware Version

2.5.18.89ebafc

### Description

System: t-beam with latest firmware
Powering with battery and charging with solar panel over USB

For meshtastic a good node coverage is crucial and there fore exposed solar powered nodes placed are often physical hard accessible. Once the AXP2101 chip detects a low battery the MCU puts the device into deep sleep! 
here is the place in the code:
https://github.com/meshtastic/firmware/blob/cdcbf4c61550e45c125e17a20aff4275e9389655/src/PowerFSM.cpp#L65

Unfortunately only pressing a button let the node resume to life. Pushing RST resets the device and pushing the GPIO38 button wakes from deep sleep and lets the device resume to power.

Repowering over USB DOES NOT exit the deep sleep state, since interrups  from the AXP are ignored in firmware:
https://github.com/meshtastic/firmware/blob/cdcbf4c61550e45c125e17a20aff4275e9389655/src/Power.cpp#L776

So once one the device is in deep sleep you have to drive to the node and push a button, this imho contradicts the spirit of meshtastic and self-sufficient systems.

Suggestion:
Shutting the device always down on client mode! Then a repowering of the USB brings the device back to life...

fifieldt, 2025-01-07:
Thanks for the report.

After reading, I'm not sure what we can do in firmware to assist this. It might be simpler to solve in hardware....

mentanah, 2025-01-07:
Why Not schutting off the device in lowBattSDSEnter() as in shutdownEnter()?

Ccz119, 2025-02-08:
Maybe a way to be able to turn off the deep sleep? Would be useful. Why 0 is one mimute?

gustavoulyssea, 2025-02-25:
I am dealing with the same issue with a device placed on the roof of a building powered by a small solar panel. Physical access to the device is very hard and depends on other people. I was wondering If the device could not go to this deep power saving state, or if it could read the battery state and resume when battery reaches let us say 50% or any other value.

Ccz119, 2025-03-26:
Is a way to turn off the deep sleep? Have
Ing all other enabled?

closed 2025-05-18 by github-actions[bot]