meshtastic/firmware #5796: [Feature Request]:

lpspot, 2025-01-09:
### Platform

Linux Native

### Description

Can install the arm64 (raspberry pi 2) on Ubuntu 24.0,
The libyaml-cpp0.7 in not ubuntu noble but libyamlcpp0. 8 is.

Apt get-install ./meshtasticd_2.5.18.89ebafc_armhf.deb
 Throws:
The following packages have unmet dependencies:
 meshtasticd : Depends: libyaml-cpp0.7 but it is not installable
E: Unable to correct problems, you have held broken packages.
 Thanks!

vidplace7, 2025-01-09:
This will be addressed by our new packaging systems, due to be announced with 2.6.x.

@lpspot For now I'd recommend that you try the *testing* builds located here:
https://launchpad.net/~meshtastic/+archive/ubuntu/daily

Later, additional repos will be populated for beta and alpha releases, but for now only `daily` (bleeding edge) builds are provided.

```shell
sudo apt update -y && sudo apt install software-properties-common
sudo add-apt-repository ppa:meshtastic/daily
sudo apt update -y
sudo apt install meshtasticd
```

closed 2025-01-09 by thebentern

lpspot, 2025-01-10:
> This will be addressed by our new Ubuntu/Debian packaging systems, due to be announced with 2.6.x.
> 
> @lpspot For now I'd recommend that you try the Ubuntu _testing_ builds located here: https://launchpad.net/~meshtastic/+archive/ubuntu/daily
> 
> Later, additional repos will be populated for beta and alpha releases, but for now only `daily` (bleeding edge) builds are provided.
> 
> ```shell
> sudo apt update -y
> sudo apt install -y software-properties-common
> 
> sudo add-apt-repository ppa:meshtastic/daily
> sudo apt update -y
> sudo apt install meshtasticd
> ```

Works perfectly! Thank you for helping and congrats for your contribution for this great project,!