meshtastic/firmware #5802: [Bug]: Lilygo T3S3 E-Paper - I2C Pin definition for environment data

chixxi, 2025-01-09:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

2.5.15

### Description

I was unable to make a INA219 environment sensor working on the  Lilygo T3S3 E-Paper, always got the message "Environment No measurement" on the display. Sensor is connected to the Qwiic connector on the PCB (RC43 / RC44 / GND / 3.3V).

Base on knowing that this works OLED T3S3 I did some research. If you add the two lines below to https://github.com/meshtastic/firmware/blob/master/variants/tlora_t3s3_epaper/variant.h and build Meshtastic, the environment measurements start to work:
`
#define I2C_SDA1 43
#define I2C_SCL1 44
`

Not committing cause I don't have a GPS module to test the impact. In the same file the definition of the GPS pins might be conflicting:
`
#define GPS_RX_PIN 44
#define GPS_TX_PIN 43
`

chixxi, 2025-01-09:
I'll test the already defined Pin 12 and 18 tomorrow, didn't see that before diving in. Probably it is enough to document that somewhere.

closed 2025-03-19 by github-actions[bot]

haidanvlad, 2025-08-25:
I'm facing the same issue with BME280, any ideas how it would work? I don't have a GPS and not interested in having one but doesn't work via qwiic anyway .