meshtastic/firmware #5816: [Bug]: T1000-E critical error & crash

cracky22, 2025-01-11:
### Category

Hardware Compatibility

### Hardware

Seeed Card Tracker T1000-E

### Firmware Version

v2.5.18.89ebafc

### Description

The Seed T1000-E Tracker for meshtastic crashed after some time with the error. I have attached logs for you below. please fix this

this is similar to #5491

[meshtastic-log-2025-01-11T11-08-05.897Z.log](https://github.com/user-attachments/files/18385518/meshtastic-log-2025-01-11T11-08-05.897Z.log)

[meshtastic-log-2025-01-11T09-57-08.007Z.log](https://github.com/user-attachments/files/18385519/meshtastic-log-2025-01-11T09-57-08.007Z.log)

[meshtastic-log-2025-01-06T11-36-45.749Z.log](https://github.com/user-attachments/files/18385520/meshtastic-log-2025-01-06T11-36-45.749Z.log)

[meshtastic-log-2025-01-06T10-17-18.275Z.log](https://github.com/user-attachments/files/18385523/meshtastic-log-2025-01-06T10-17-18.275Z.log)

[meshtastic-log-2025-01-06T10-05-14.999Z.log](https://github.com/user-attachments/files/18385524/meshtastic-log-2025-01-06T10-05-14.999Z.log)

[meshtastic-log-2025-01-06T09-21-21.705Z.log](https://github.com/user-attachments/files/18385525/meshtastic-log-2025-01-06T09-21-21.705Z.log)

### Relevant log output

```Shell
WARN | 12:06:53 3639 [RadioIf] Can not send yet, busyTx
ERROR | 12:06:53 3639 [RadioIf] Hardware Failure! busyTx for more than 60s
ERROR | 12:06:53 3639 [RadioIf] NOTE! Record critical error 8 at src/mesh/RadioLibInterface.cpp:100
```

cracky22, 2025-01-11:
```
INFO | 12:23:07 305 Tell client we have new packets 60
INFO | 12:23:07 305 BLE notify fromNum
WARN | 12:23:07 305 [RadioIf] Can not send yet, busyTx
WARN | 12:23:07 305 [RadioIf] Can not send yet, busyTx
WARN | 12:23:07 305 [RadioIf] Can not send yet, busyTx
DEBUG | 12:23:07 305 FromRadio=STATE_SEND_PACKETS
WARN | 12:23:07 305 [RadioIf] Can not send yet, busyTx
WARN | 12:23:07 305 [RadioIf] Can not send yet, busyTx
DEBUG | 12:23:07 305 FromRadio=STATE_SEND_PACKETS
DEBUG | 12:23:07 305 phone downloaded packet (id=0x1c9df89e fr=0x09b65788 to=0xffffffff, WantAck=0, HopLim=7 Ch=0x0 Portnum=3 rxtime=1736594586 hopStart=7 priority=10)
WARN | 12:23:07 305 [RadioIf] Can not send yet, busyTx
WARN | 12:23:07 305 [RadioIf] Can not send yet, busyTx
WARN | 12:23:07 305 [RadioIf] Can not send yet, busyTx
DEBUG | 12:23:07 306 [GPS] $PAIR001,021,0*38..$PAIR021,AG3335M_V2.6.0.AG3335_20221230,S,I,cfced9e,2212300120,a2b7,0,,,735e4e5a,2212300107,e951622,2212300116,,,,,-15.48,-15.48,-14.02,-15
WARN | 12:23:07 306 [RadioIf] Can not send yet, busyTx
WARN | 12:23:07 306 [RadioIf] Can not send yet, busyTx
WARN | 12:23:07 306 [RadioIf] Can not send yet, busyTx
WARN | 12:23:07 306 [RadioIf] Can not send yet, busyTx
WARN | 12:23:07 306 [RadioIf] Can not send yet, busyTx
WARN | 12:23:07 306 [RadioIf] Can not send yet, busyTx
WARN | 12:23:07 306 [RadioIf] Can not send yet, busyTx
WARN | 12:23:08 306 [RadioIf] Can not send yet, busyTx
WARN | 12:23:08 306 [RadioIf] Can not send yet, busyTx
WARN | 12:23:08 307 [RadioIf] Can not send yet, busyTx
INFO | 12:23:39 338 [RadioIf] toRadioWriteCb data 0x20015ffa, len 51
DEBUG | 12:23:39 338 [RadioIf] New ToRadio packet
DEBUG | 12:23:39 338 [RadioIf] PACKET FROM PHONE (id=0xd69bc4cb fr=0x00000000 to=0x09b65788, WantAck=0, HopLim=7 Ch=0x0 Portnum=3 priority=10)
DEBUG | 12:23:39 338 [RadioIf] Enqueued local (id=0xd69bc4cb fr=0x00000000 to=0x09b65788, WantAck=0, HopLim=7 Ch=0x0 Portnum=3 rxtime=1736594619 priority=10)
ERROR | 12:23:39 338 [RadioIf] Hardware Failure! busyTx for more than 60s
ERROR | 12:23:39 338 [RadioIf] NOTE! Record critical error 8 at src/mesh/RadioLibInterface.cpp:100
```

GUVWAF, 2025-01-11:
Thanks for the report, can you try if it is fixed with #5820?

cracky22, 2025-01-11:
@GUVWAF do you have a .uf2 file for me for the t1000e?

GUVWAF, 2025-01-11:
@cracky22 It's in the attached .zip file.
[firmware_PR5820.zip](https://github.com/user-attachments/files/18385979/firmware_PR5820.zip)

cracky22, 2025-01-11:
okay, thank you. it looks like it's fixed. is it then also fixed for #5491 at rak?

GUVWAF, 2025-01-11:
Thanks, good to hear that. You can try it on a RAK with: 
[firmware-nrf52840-rak4631-2.5.19.812aa35.zip](https://github.com/user-attachments/files/18386203/firmware-nrf52840-rak4631-2.5.19.812aa35.zip)

However, in #5491 it looks like people are experiencing a different issue related to the file system also. Let's keep that one open for now.

closed 2025-01-11 by GUVWAF

cracky22, 2025-01-12:
> Thanks, good to hear that. You can try it on a RAK with: [...]

thanks for the firmware but this is unfortunately not fixed on the rak... my rak crashes after 6-9 hours
#5491

GUVWAF, 2025-01-12:
@cracky22 With the same issue as here (busy Tx for 60 seconds)?

cracky22, 2025-01-12:
@GUVWAF Unfortunately I can't log it because I don't have a USB connection to the node. Do you have a way that I can log the data remotely or that it can be saved in the flash?

GUVWAF, 2025-01-12:
You can use rsyslog, but only if you have an Ethernet (or WiFi) connection.

cracky22, 2025-01-12:
Nope i dont have wifi on my rak...

cracky22, 2025-01-12:
My Problem is, inside the House i am Not alle to get 80+ nodes and outsoide on i dont have usb and a Computer to log...

garthvh, 2025-01-13:
Use the client proxy to connect to mqtt and you will have 80 nodes in a hour