meshtastic/firmware #5817: [Bug]: shutdown fail

cracky22, 2025-01-11:
### Category

Other

### Hardware

Seeed Card Tracker T1000-E

### Firmware Version

v2.5.18.89ebafc

### Description

[meshtastic-log-2025-01-11T11-27-01.114Z.log](https://github.com/user-attachments/files/18385575/meshtastic-log-2025-01-11T11-27-01.114Z.log)

### Relevant log output

```Shell
DEBUG | 12:26:52 192 [Button] Radio wait to sleep, txEmpty=0

DEBUG | 12:26:52 192 [Button] Radio wait to sleep, txEmpty=0

DEBUG | 12:26:52 192 [Button] Radio wait to sleep, txEmpty=0

DEBUG | 12:26:52 192 [Button] Radio wait to sleep, txEmpty=0

DEBUG | 12:26:52 192 [Button] Radio wait to sleep, txEmpty=0

DEBUG | 12:26:52 192 [Button] Radio wait to sleep, txEmpty=0

DEBUG | 12:26:52 193 [Button] Radio wait to sleep, txEmpty=0

DEBUG | 12:26:52 193 [Button] Radio wait to sleep, txEmpty=0

DEBUG | 12:26:52 193 [Button] Radio wait to sleep, txEmpty=0

DEBUG | 12:26:53 193 [Button] Radio wait to sleep, txEmpty=0

DEBUG | 12:26:53 193 [Button] Radio wait to sleep, txEmpty=0

DEBUG | 12:26:53 193 [Button] Radio wait to sleep, txEmpty=0

ERROR | 12:26:53 193 [Button] NOTE! Record critical error 2 at src/sleep.cpp:176

ERROR | 12:26:53 193 [Button] assert failed src/sleep.cpp: 177, void waitEnterSleep(bool), te
```

cracky22, 2025-01-11:
[meshtastic-log-2025-01-11T11-33-06.074Z.log](https://github.com/user-attachments/files/18385593/meshtastic-log-2025-01-11T11-33-06.074Z.log)

GUVWAF, 2025-01-11:
I believe this is because of this: 

https://github.com/meshtastic/firmware/blob/e7802d960fac813e58e6b0e8ed085d91801e5dd8/src/sleep.cpp#L171-L172

This will not work because during `delay()`, ArduinoThread will not call the `NotifiedWorkerThread` of RadioLibInterface and thus it won't reach `TRANSMIT_DELAY_COMPLETED`, keeping the Tx queue non-empty. We'd need to find a way to fix this.

cracky22, 2025-01-11:
log from t1000e with testbuild 2.5.19.6f0ab0f60
[meshtastic-log-2025-01-11T14-39-41.525Z.log](https://github.com/user-attachments/files/18386175/meshtastic-log-2025-01-11T14-39-41.525Z.log)

mverch67, 2025-09-01:
fixed by [#7792](https://github.com/meshtastic/firmware/pull/7792)

closed 2025-09-01 by mverch67