meshtastic/firmware #5823: [Bug]: T114 shows battery percentage when no battery connected

Pigpog, 2025-01-12:
### Category

Hardware Compatibility

### Hardware

Heltec Mesh Node T114

### Firmware Version

2.5.12.aa184e6

### Description

When no battery is connected, the T114 displays a seemingly-random battery charge percentage. This value is sent with telemetry as well. It never acts as if it is connected to a power supply.

todd-herbert, 2025-01-12:
Same with some older Heltec boards too like Wireless Paper. I spent quite a while trying to come up with some clever way to detect whether a battery was connected or not, but never had any luck.

closed 2025-01-12 by thebentern

Pigpog, 2025-01-12:
Looks like it's just a spotty USB-C connection. Moving the cable around flickers the orange LED. Trying with a better cable, the device still shows a battery voltage and percentage, but at least it says it's plugged in as well.